bambulab/BambuStudio #2087: Apply dynamic slowdowns on overhangs more effectively

psiberfunk, 2023-07-20:
Prusaslicer, in Beta 1 of 2.60 introduced a meaningful change that increases print quality for overhangs by more intelligently applying overhang slowdowns as noted here:   https://github.com/prusa3d/PrusaSlicer/releases 

Specifically [FR] https://github.com/prusa3d/PrusaSlicer/issues/9756 and this commit appear to be the change:

https://github.com/prusa3d/PrusaSlicer/commit/3e3ccc200e703fa6b21a610a8c828be23b301873

Integrating this into bambu slicer would improve overhang curling performance at minimal speed cost and should probably be the default.  Please consider pulling this into the slicer.

SaltWei, 2023-07-21:
Could you raise an example about how effective and how it does？

psiberfunk, 2023-07-21:
> Could you raise an example about how effective and how it does？

That's already in the [FR] on the prusaslicer branch ?  A standard Benchy will show you the difference.

SaltWei, 2023-07-21:
@QingZhangBambu Please evaluate

davidjdixon, 2025-09-17:
Could we please get this issue on the radar again? Orca Slicer has a setting for 'Slow down for curled perimeters' and this setting is invaluable to delicate and complicated perimeters (eg in vase mode) where the perimeters tend to curl slightly on small overhangs. This hugely reduces the occurrence of nozzle collisions with the perimeters, and I'm currently forced to use Orca Slicer for certain models purely for this feature.