bambulab/BambuStudio #2088: P1S MicroSD Card fails to load in Studio

jhandel, 2023-07-20:
### Bambu Studio Version

1.7.0.65

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

Microsoft Surface Studio Laptop
32gb memory 
Nvidia Graphics
i7 Processor

### Printer

Bambu Labs P1s

### How to reproduce

Log into Bambu Labs account that has a P1S with firmware 1.03.00.00
Connect to P1s - 
Click on "Micro SD Card"

### Actual results

Receive the message "Initialize Failed (Not supported on the current printer version)!" 

### Expected results

Be able to see list of printable files on the SD card.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12113966/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

zachrdunn, 2023-07-22:
Same. Also prohibits you from viewing timelapse files.

mrponzo, 2023-07-22:
have the same issue with the new P1S

bambu123, 2023-07-23:
It is under develop for p1s.

zachrdunn, 2023-07-28:
Is there a general timeline for this? Curious about this as well as getting a P1S option in the stable version of bambu studio

Bernstea, 2023-08-03:
just installed 1.7.1.62 on Mac. Same issues - Receive the message "Initialize Failed (Not supported on the current printer version)!, no files, no time-lapse? When can we expect fix/update release?  Thanks!

PedroLageTavares, 2023-08-03:
+1 with the same issue and looking forward for the fix.

Quite curious about the tag changing from "Bug" into "Feature". How is this issue considered an feature?

nweishar, 2023-08-07:
+1 Same issue for P1S with latest firmware 01.04.00.00. Please provide a fix!

gpoleze, 2023-08-08:
If this "feature" was in place, I would be able to say that I 100% in love with my P1S. With it lacking, it feels half-baked.

simdeft, 2023-08-09:
Same issue.
P1S with firmware 1.03.00.00
Studio 1.7.3.50

Smeltie, 2023-08-10:
Count me in, same "Issue" here.
Latest version as of 10th of august.

Briggsy92, 2023-08-11:
Still the same issue with P1S firmware 01.04.00.00 and Bambu Studio version 1.7.3.50.

dalealleshouse, 2023-08-11:
I also have the same issue.
P1P Version 01.04.00.00

bergby, 2023-08-12:
Same issue, P1S , 1.04, 1.7.3.50.

truenorthzero, 2023-08-13:
Also experiencing the same issue with P1S firmware - 01.04.00.00, Bambu Studio ver - 1.7.3.50.

rossbeck, 2023-08-13:
> If this "feature" was in place, I would be able to say that I 100% in love with my P1S. With it lacking, it feels half-baked.

This feature was literally added less than a week ago and if you'd read the notes, it's not implemented for P1P/P1S yet. Patience. Many of us P1P owners have lived just fine without it for a year now.

porterble, 2023-08-16:
After the cloud failure this is now a critical issue as LAN mode is intrinsically broken on the P1S with firmware 01.04.00.00 and Bambu Studio 1.7.3.50. There is no way to print using the AMS from Bambu Studio to the P1S without using the cloud.

IceManJPx, 2023-08-29:
Same issue 
Firmware 01.04.00.00
BambuStudio 1.7.4.52

Feels like i got the Kickstarter edition ...

hsavior, 2023-08-31:
Same here. 

Firmware 01.04.00.00
BambuStudio 1.7.4.52

vashspike2, 2023-09-05:
> It is under develop for p1s.
> […](#)

Thanks @bambu123 , do you know roughly when this feature will come to P1S?

bambu123, 2023-09-05:
> > It is under develop for p1s.
> > […](#)
> 
> Thanks @bambu123 , do you know roughly when this feature will come to P1S?

@sodachen22 will schedule this feature.

sodachen22, 2023-09-05:
Browsing 3MF on the SD card on the P1 series printer is still in development.

djzinc, 2023-09-21:
Same here.
Firmware 01.04.00.00
BambuStudio 1.7.4.52
Workaround for me is to read the mSD card in a computer.
There is a folder named "timelapse" with the *.avi files.

druckgott, 2023-09-24:
same Here 
Firmware 01.04.00.00
BambuStudio 1.7.6.92

Workaround is to connect with **FTP**: PI from Printer, **User** bblp, **PW**, Password is the access code found on the LCD in the WiFi settings, **Port** 990

Erre404, 2023-09-28:
Same problem
Firmware 01.04.01.00
BambuStudio 1.7.6.92

RyanEwen, 2023-09-28:
+1, have been waiting for this feature since Feb, as well as some way to re-print using Bambu Studio instead of the app or device itself,

Craftzman7, 2023-09-29:
@bambu123 any update on this?

f8al, 2023-10-02:
+1 on this issue

Puresoft, 2023-10-02:
+1 for me too. 
Is the P1S that much different from the P1P?

hsavior, 2023-10-02:
The P1S is the exact same but with all accessories (camera, light, panels)

mrtwebdesign, 2023-10-04:
> Same problem Firmware 01.04.01.00 BambuStudio 1.7.6.92

My P1S setup is the same as above.
Rule of thumb: Never count on software features that are not already available when you purchase a product.
We may never get this feature if they aren't prioritizing it.

I'm not optimistic, I'm a software developer.

FlyingT, 2023-10-05:
> +1, have been waiting for this feature since Feb, as well as some way to re-print using Bambu Studio instead of the app or device itself,

The P1S started shipping in July, dont be overly dramatic

mike391, 2023-10-05:
> +1 for me too. 
> Is the P1S that much different from the P1P?

It's the P1P with sides, cable chain and a chamber fan as well as some slight revisions to the movement system for durability. It's essentially the same machine though, newer p1ps may even have the movement system revisions as well

matt-gm, 2023-10-15:
+1

Firmware: 01.04.01.00
BambuStudio: 1.7.7.89

CountOlaf87, 2023-10-16:
+1 for me

b-pardeep, 2023-10-23:
Would love a general timeline on this. Thanks.

gsingh93, 2023-11-07:
All of the +1s are not helpful. Just click the thumbs up react in the original post and subscribe to the issue. Leave the comments for actual discussion about the issue.

b-pardeep, 2023-11-07:
> All of the +1s are not helpful. Just click the thumbs up react in the original post and subscribe to the issue. Leave the comments for actual discussion about the issue.

Shukria bhai ji for the suggestion!!

yezhipeng2417, 2023-11-17:
It doesn't work on A1 mini as well

Meth0d007, 2023-11-27:
It's quite sad, not able to manage or even delete files from SD via Remote feels ancient even compared to the very ancient WebUI from my Prusa Mini+

Klages, 2023-11-27:
It will be available, probably with the next release: https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.57

[image]

raster, 2023-11-27:
> Am i the only one who has to pull and wipe his SD because it got cluttered with different test-versions of prints? Looking forward to a delete from SD function.

I just clear things out via FTP.

felixkriechbaum, 2023-12-06:
> It will be available, probably with the next release: https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.57
> 
> ![firefox_ydyTs2jlke](https://private-user-images.githubusercontent.com/3393617/285919529-61f865be-0d56-4673-a244-54ff8a1c2f68.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE4NjIxOTYsIm5iZiI6MTcwMTg2MTg5NiwicGF0aCI6Ii8zMzkzNjE3LzI4NTkxOTUyOS02MWY4NjViZS0wZDU2LTQ2NzMtYTI0NC01NGZmOGExYzJmNjgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMDZUMTEyNDU2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZjA3ZTc5MTRmOTVlMjhjYjgzMTA3M2IwNWQzMThlMzI0MTRiZjU3ODE4YWYxOWI1NzE5NDhlZTY1YjYxMDRmMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.tf_tqu_mIPn-r5PKD-LVImhvhLxHxOjpePF3fgqN3Lw)

Tried to do it with BETA 3 and still the same message

[image]

Zaaap, 2023-12-06:
> > It will be available, probably with the next release: https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.57
> > ![firefox_ydyTs2jlke](https://private-user-images.githubusercontent.com/3393617/285919529-61f865be-0d56-4673-a244-54ff8a1c2f68.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE4NjIxOTYsIm5iZiI6MTcwMTg2MTg5NiwicGF0aCI6Ii8zMzkzNjE3LzI4NTkxOTUyOS02MWY4NjViZS0wZDU2LTQ2NzMtYTI0NC01NGZmOGExYzJmNjgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMDZUMTEyNDU2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZjA3ZTc5MTRmOTVlMjhjYjgzMTA3M2IwNWQzMThlMzI0MTRiZjU3ODE4YWYxOWI1NzE5NDhlZTY1YjYxMDRmMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.tf_tqu_mIPn-r5PKD-LVImhvhLxHxOjpePF3fgqN3Lw)
> 
> Tried it with BETA 3 and still the same message
> 
Please follow your own link and read:
>Please note that this feature relies on the printer firmware being upgraded to a new version(P1P/P1S v1.5 and A1 mini v1.2 to be released).
Do you have that Firmware update?
No? Then wait patiently. 
Or use FTP in the meantime.

CreativeMechanics, 2023-12-10:
SAME. I just got my P1S today, updated the firmware, and downloaded the software. I'm getting this error message too. :(

CreativeMechanics, 2023-12-10:
> > Am i the only one who has to pull and wipe his SD because it got cluttered with different test-versions of prints? Looking forward to a delete from SD function.
> 
> I just clear things out via FTP.

I'm new to all of this. What is FTP? How does one access it?

TaylorN15, 2023-12-10:
> SAME. I just got my P1S today, updated the firmware, and downloaded the software. I'm getting this error message too. :(

This is not a forum, that being said, look through all the other comments before making a new comment. This has been asked and answered. The firmware update has not yet been released, the Bambu Studio version is still in beta anyway. 

If you don’t know what FTP Is, then I suggest you start with a simple Google search.

CreativeMechanics, 2023-12-10:
> This is not a forum, that being said, look through all the other comments before making a new comment. This has been asked and answered. The firmware update has not yet been released, the Bambu [...]

Google Search: "FTP: File Transfer Protocol." Hmmm... that doesn't seem to help. 
Do you just mean taking the SD card out of the printer and putting it into the computer? You could have just said that if that's what you meant.

TaylorN15, 2023-12-10:
> Google Search: "FTP: File Transfer Protocol." Hmmm... that doesn't seem to help. Do you just mean taking the SD card out of the printer and putting it into the computer? You could have just said that if that's what you meant.

I meant to search for something like “Bambu P1S FTP” -
[Here](https://forum.bambulab.com/t/we-can-now-connect-to-ftp-on-the-p1p/6464/27) I’ve done it for you

zachrdunn, 2023-12-10:
> I meant to search for something like “Bambu P1S FTP” -

Just a reminder. This is not a forum,

zesteph, 2023-12-15:
> It will be available, probably with the next release: https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.57 [...]

hi
i m new user from bambu lab p1s and same problem
Firmware: 01.04.02.00 (Last)
BambuStudio: 1.8.2.56 (Last no beta and more high from you screen)

TaylorN15, 2023-12-15:
> hi i m new user from bambu lab p1s and same problem Firmware: [01.04.02.00](tel:01.04.02.00) (Last) BambuStudio: 1.8.2.56 (Last no beta and more high from you screen)

The new firmware version has not been released yet… read all the previous comments.

cqmiao, 2023-12-28:
My A1 mini is same problem
Firmware: 01.01.03.00 (Last)
BambuStudio: 1.8.2.56 (offical version)
[image]

gsingh93, 2024-01-02:
On the latest version of Bambu Studio (1.8.2.56) with the latest firmware on my P1S (01.05.00.00), I can now see files on the SD card. This issue should be closed at this point, A1/A1 mini users should open a separate issue if they still having problems.

lloydbayley, 2024-01-03:
Can confirm it works here too with those versions. It's great to be able to work with the models from the studio. Is it correct that downloading a time-lapse is painfully-slow? Anyone else?