bambulab/BambuStudio #2091: Video stopped and not possible to renew

eXenInfo, 2023-07-21:
### Bambu Studio Version

1.7.0.65

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10 Home

### Additional system information

Intel Core i5-4310U
16 GB RAM

### Printer

Bambu Lab P1P

### How to reproduce

Start an individual print
Start video on the device page.
Between 5 minutes and or later the video stream is frozen and cannot refreshing anymore.
If I start the Orca Slicer 1.6.3 on the same machine, the video is loading immediately and works well.

### Actual results

Video is frozen

### Expected results

shoud work as the normal Bambu or Orca version

### Project file & Debug log uploads

I do not have anything as asked for.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

VeryKross, 2023-07-22:
This is very consistent (and frustrating) behavior on my system as well. I have to open Handy app as a work-around for remote monitoring.

bambu123, 2023-07-24:
> This is very consistent (and frustrating)obehavior on my system as well. I have to open Handy app as a work-around for remote monitoring.

Please send me the logs of Studio.

8stated, 2023-07-27:
Log for this bug on macOS attached here.

[debug_Fri_Jul_21_10_25_27_864.log.0.zip](https://github.com/bambulab/BambuStudio/files/12188836/debug_Fri_Jul_21_10_25_27_864.log.0.zip)

bambu123, 2023-07-28:
> Log for this bug on macOS attached here.
> 
> [debug_Fri_Jul_21_10_25_27_864.log.0.zip](https://github.com/bambulab/BambuStudio/files/12188836/debug_Fri_Jul_21_10_25_27_864.log.0.zip)

This is not the log of the problem scene. Please send me all the log files if you are not sure which one is.

8stated, 2023-07-31:
[log.zip](https://github.com/bambulab/BambuStudio/files/12210539/log.zip)

bambu123, 2023-08-01:
> All my logs
> 
> [log.zip](https://github.com/bambulab/BambuStudio/files/12210539/log.zip)

What's the problem time in your log?

I found some failure in the log when connecting to the printer which may caused by watching live video by more that one client.
The P1P printer is lack of resources for tasks. For example, the liveview can't work when downloading print files for print.

bambu123, 2023-08-02:
> It would be all the logs for the past 24 hours (since I started using Bambu Studio 1.7.0.65), I upgraded my P1P to a P1S, but the camera was working fine in past BS releases for many months.

It seam your Studio (PC) and your printer are in diffirent network and the 1.6 version can't support this condition.
The 1.6 version uses local tcp connection to the printer, but the 1.7 version uses remote connection.

8stated, 2023-08-02:
It is all on the same LAN (172.16.1.xxx), the printer even uses the same IP address since it arrived (172.16.1.151) on WIFI 2.4

Any other information you need from me?

bambu123, 2023-08-02:
> It is all on the same LAN (172.16.1.xxx), the printer even uses the same IP address since it arrived (172.16.1.151) on WIFI 2.4 Any other information you need from me?

I see your problem is never restart video after a stop. It was a problem, please update the network plugin and retry.

8stated, 2023-08-02:
I deleted all the application and its preferences, reinstalled 1.7.1.62 the update with network plugin, it seems better, but not perfect.

I keep getting this warning when trying to print files on my upgraded P1P > P1S, P1S is selected as the printer and nozzle etc:

bambu123, 2023-08-03:
> I deleted all the application and its preferences, reinstalled 1.7.1.62 the update with network plugin, it seems better, but not perfect. I keep getting this warning when trying to print files on my upgraded P1P > P1S, P1S is selected as the printer and nozzle etc:

Any new logs?

bambu123, 2023-08-03:
I found it may relate to the old P1P firmware, please wait for the 01.04 version of P1P update.

8stated, 2023-08-03:
Thank makes sense, I shall hope the update is coming soon. Many thanks!

cmguo, 2023-08-10:
#2162

bambu123, 2023-08-21:
@8stated Have you the P1P 1.4 version? How is this problem now?

8stated, 2023-08-21:
I have version 1.4 and the issue still persists. 

Should I run the Network Test in developer mode?

bambu123, 2023-08-21:
> I have version 1.4 and the issue still persists. Should I run the Network Test in developer mode?
> […](#)

Please send me the app logs of Studio.

bambu123, 2023-08-25:
@VeryKross @eXenInfo How has your recent situation been?

Did you upgrade the firmware of P1P to version 1.4?

bambu123, 2023-08-25:
@VeryKross @eXenInfo @8stated Please

8stated, 2023-08-25:
Yes in my last reply to you I said I had upgraded to 1.4.  Did you see anything from the log files I sent you earlier this week?

bambu123, 2023-08-25:
> Yes in my last reply to you I said I had upgraded to 1.4.  Did you see anything from the log files I sent you earlier this week?

Sorry, I didn't see the attachment in your email.
To troubleshoot this problem, logs of the printer are also required. Please help provide logs of the printer firmware using the method provided at the link below.

https://wiki.bambulab.com/en/p1/troubleshooting/how-to-upload-log

8stated, 2023-08-25:
Ok I’m away from my printer until after 4pm. Then I can send you more logs

https://wiki.bambulab.com/en/p1/troubleshooting/how-to-upload-log

eXenInfo, 2023-08-25:
> @VeryKross @eXenInfo How has your recent situation been?
> 
> Did you upgrade the firmware of P1P to version 1.4?

Hi,
Yes I also upgraded to 1.4 and still the same as issues a 8stated described so pretty good that I have nothing to add.

8stated, 2023-08-25:
Logger file:

https://drive.google.com/file/d/1nUTOn0zD1yXkMK5GXNZIfW7BNTqGOg9D/view?usp=drive_link <https://drive.google.com/file/d/1nUTOn0zD1yXkMK5GXNZIfW7BNTqGOg9D/view?usp=drive_link>

16mb!

bambu123, 2023-08-25:
> Logger file: https://drive.google.com/file/d/1nUTOn0zD1yXkMK5GXNZIfW7BNTqGOg9D/view?usp=drive_link <https://drive.google.com/file/d/1nUTOn0zD1yXkMK5GXNZIfW7BNTqGOg9D/view?usp=drive_link> 16mb!
> 

Thank you, we will analyze it right away.

bambu123, 2024-02-06:
Fixed by the printer firmware 1.5.

bambu123, 2024-02-06:
> Fixed by the printer firmware 1.5.

closed 2024-02-06 by bambu123