bambulab/BambuStudio #2092: Apply separate patter to Topmost layer.

randyre, 2023-07-21:
Would like to see the ability to apply a surface pattern to the topmost layer separate from whatever the rest of the print has selected

**Describe the solution you'd like**
Provide a selection box or drop-down menu

[image]

QingZhangBambu, 2023-08-02:
you can use a modifier to change the pattern.
[image]

randyre, 2023-08-05:
@QingZhangBambu 

That's great for simple objects, but for complex objects it's hard to do. Plus Cura does it, would be nice not to have to reinvent the wheel

QingZhangBambu, 2023-08-09:
ok, I would add that to our feature pool. and the concerned staff will discuss that.

closed 2024-04-07 by randyre