bambulab/BambuStudio #2094: Can't set chamber fan at desired speed while printing

michelem09, 2023-07-21:
### Bambu Studio Version

1.7.0 beta 2

### Where is the application from?

Bambu Lab github releases

### OS version

MacOS 13.4.1

### Printer

Bambu Lab P1P

### How to reproduce

1. in the gcode there are conditional sentences to enable chamber fan (M106 P3) to some values like this the Bambu PLA-CF preset
2. start a print
3. go to device and try to change the chamber fan

I attached a screencast

https://github.com/bambulab/BambuStudio/assets/439555/fa7c0f78-14b8-44a9-8546-ff5bffb68b5f

### Actual results

The fan won't change and the value comes back to the preset one (from gcode conditions)

### Expected results

I expect to be able to change chamber fan value by 10% as per UI

### Project file & Debug log uploads

It's not project related it's systematic with any print

### Checklist of files to include

- [ ] Log file
- [ ] Project file

michelem09, 2023-07-26:
I found another weird behaviour related to fans, try also this:

1. While printer is in stand-by enable the Part fan **from the printer display**
2. Now go to BambuStudio and try to disable the Part fan
3. The fan won't stop (you need to stop it from the printer display)

walterwongbbl, 2023-08-23:
Hi. @michelem09 
What is the firmware version number of your printer？

michelem09, 2023-08-23:
I updated to latest one 01.04.00.00 and everything listed here seems to work now. Thanks!

closed 2023-08-23 by michelem09