bambulab/BambuStudio #2096: Skip AMS Blacklist check??

Fusiable, 2023-07-21:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
What is this "Skip AMS blacklist check" option under the Bambu Studio preference list?

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

RyanEwen, 2023-07-27:
I've been dying to know what this setting does as well

antoan-boss, 2023-08-10:
Anyone knows what it does?

antoan-boss, 2023-08-10:
It skips the check of filaments in AMS based file on rules in file below:
[](https://github.com/bambulab/BambuStudio/blob/f14b6252e384bdd86075d3f18c7f4b53bae08a51/resources/printers/filaments_blacklist.json)

tsmith35, 2023-08-17:
@antoan-boss thanks for the pointer!

AbeFM, 2023-10-04:
Well isn't that good to know.
I've been begging for a way to not have to lie to my machine for..... well, since a month after I got the machine and it stopped working.

---
I edited out the TPU restrictions, but the AMS itself still complains. And I can't set it in the device settings in the slicer, either.
Still, thanks. :-)

GuzziRaz, 2024-01-25:
So what is needed to close this issue is merely a descriptive tooltip.

daniel-brenot, 2024-10-06:
> Well isn't that good to know. I've been begging for a way to not have to lie to my machine for..... well, since a month after I got the machine and it stopped working.
>
> I edited out the TPU [...]

So usually this feature would do exactly what you think, but for some reason they decided to go the extra mile and HARD CODE in any TPU filaments as not being allowed to print. This is insanely annoying and my solution to this is that i'm recompiling their code with those lines removed because I refuse to lie to my printer.

tsmith35, 2024-10-06:
> extra mile and HARD CODE in any TPU filaments as not being allowed to print. This is insanely annoying and my solution to this is that i'm recompiling their code with those lines removed because I refuse to lie to my printer.

Which files / lines do you edit?

daniel-brenot, 2024-10-06:
> > extra mile and HARD CODE in any TPU filaments as not being allowed to print. This is insanely annoying and my solution to this is that i'm recompiling their code with those lines removed because I refuse to lie to my printer.
> 
> Which files / lines do you edit?

I've got a fork that shows how I did it. There's basically just 2 files, one with an if statement that seems to be to check in one point and another to throw the error during blacklist checking. Honestly they should move the code for blacklist checking to make sure that it doesn't trigger if the checkbox is checked for disabling it, but I have a feeling this is completely intentional to prevent support requests from people who ignore instructions. I'm personally happy to void my warranty as I need to be able to mix tpu and petg in a print right now and the AMS is the only way for me to do that.

daniel-brenot, 2024-10-06:
@tsmith35 The most recent windows build on my fork seems to work. Feel free to download the exe file and try it.

AbeFM, 2024-10-07:
I'm in on this.  I'm short on sleep right now because I stayed up last
night manually handling rigid tpu and cursing the name of whomever put this
foot down at Bambu

AbeFM, 2024-10-07:
Daniel,
I wonder if it makes any sense to air this in front of the Orca Slicer
guys. They do have a certain penchant for actually doing stuff just because
it makes sense. I'm starting to think years of begging and thinking
tomorrow they will change their minds is a little too fool-me-once.

daniel-brenot, 2024-10-07:
> Daniel, I wonder if it makes any sense to air this in front of the Orca Slicer guys. They do have a certain penchant for actually doing stuff just because it makes sense. I'm starting to think years of begging and thinking tomorrow they will change their minds is a little too fool-me-once.
> […](#)

Maybe, although I believe these changes were mainly made on the bambu studio end, not on the orca slicer end, so I fail to see how it will help.

I went to go chat about this in the discord, but it seems the official discord bans talking about modifying your printer, which is a strange choice.

AbeFM, 2024-10-07:
Ok, tired of beating my face against the wall on this one, Bambu. You could have at least given a reason.

Posted to Orca, let's hope saner heads prevail here. https://github.com/SoftFever/OrcaSlicer/issues/7045

tsmith35, 2024-10-14:
> @tsmith35 The most recent windows build on my fork seems to work. Feel free to download the exe file and try it.

Sorry I haven't replied. My dad died on 10/6. I'll check it out soon.

AbeFM, 2024-10-14:
Take all the time you need.  Printing will wait.

MistressPlague, 2024-10-16:
Why doesn't this bypass the internal TPU check? This seems a bug. Skip AMS blacklist checks should fully allow any filament type in AMS.

daniel-brenot, 2024-10-16:
> Why doesn't this bypass the internal TPU check? This seems a bug. Skip AMS blacklist checks should fully allow any filament type in AMS.

After looking at the code I'm not sure if this was intentional or not. I believe they just missed a bit of code at the part where the blacklist check is done.

For some perspective, there is a function used for checking blacklists, but it does not check if the checkbox is checked because it is used by many different parts of the software. Because of this, the skip blacklist check has to be done at the specific points where the blacklist check function is called.

AbeFM, 2024-10-17:
What I can tell you is BBL is well aware of the issue.  If you get a
response, and I've tried a fair amount,  it's along the lines of TPU isn't
recommended. More power to you if you can get them to listen but I brought
it here because I feel that getting BBL to help is not going to happen.  I
can link to a few threads on discord,  Bambu forums, github, etc.  They'll
reply once,  so they are aware.  But no questions were answered,  showing
them successful prints etc got no response.

I wish I had more to add on the logic/ coding end. It worked worked the
machine was new.

phiten, 2024-10-18:
I fixed that in this PR https://github.com/SoftFever/OrcaSlicer/pull/7161 

Installers and stuff can be found in my repos actions https://github.com/phiten/OrcaSlicer/actions/runs/11401874109

Basically the PR checks if the checkbox is set and changes prohibited filaments to filament with warning so that the confirm button actually sends the print.