bambulab/BambuStudio #2098: Multiple filament slots does not change the actual filament settings

Naninani, 2023-07-22:
### Bambu Studio Version

1.7.0

### Where is the application from?

Bambu Lab github releases

### OS version

macOS Ventura 13.4.1

### Printer

P1P

### How to reproduce

1. Backup all existing profiles.
2. Wipe all profiles.
3. Logout if logged in (issues repo both logged in or out)
4. Uninstall Orca.
5. Install again.
6. Select Generic PETG.
7. Edit the Settings Override of Filament settings.
8. Set Z hop type to Slope
9. Set the Wipe Distance to 10mm
10. Save present to new one with name 1.
11. Add Primitive Cube
12. Add additional filament slot.
13. In slot 1 chose filament 1, in slot 2 chose Generic PETG
14. Slice with 1 and enable the Travel and Wipe to see the lines.
15. Set to slot 2 with the Generic PETG and slice again.
16. What happens is that it does not actually switch to slot 2 settings, but choses the settings from slot 1.
17. This happens only if there are more than 2 added filament slots. If you chose Generic PETG from slot 1, the settings are correctly updated.

### Actual results

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Expected results

It should correctly update the settings for the different filament slots.
This is an extremely nasty bug that I managed to reproduces in both Orca and Bambu slicers. It actually ruined tons of prints for me and I spent countless hours figuring out why my prints are so off when using different filament slots.

### Project file & Debug log uploads

just add simple primitive object to test

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Naninani, 2023-07-22:
cross created in Orca as well https://github.com/SoftFever/OrcaSlicer/issues/1570

RyanEwen, 2023-07-27:
This is a great find and might contribute to a bunch of other issues that people are having (myself included)

Naninani, 2023-07-27:
> This is a great find and might contribute to a bunch of other issues that people are having (myself included)

@RyanEwen are you able to repo it on your end?
I tested on different machines and 100% repo, but I want to make sure others can repo as well.

Naninani, 2023-08-16:
@zhimin-zeng-bambulab, any update on this? To me its a critical bug that is causing profiles mixing and alot of print issues.

zhimin-zeng-bambulab, 2023-08-21:
We are handling BUG on github, we will fix this BUG as soon as possible.

Naninani, 2023-10-10:
@SaltWei, @zhimin-zeng-bambulab any update on this?

zhimin-zeng-bambulab, 2023-10-10:
@Naninani  It has been fixed in V1.7.6, you can have a try.

closed 2023-10-13 by zhimin-zeng-bambulab