bambulab/BambuStudio #2099: Go Live Missing on v1.7.0.65 Public Beta 2 for P1S

rinytin, 2023-07-22:
### Bambu Studio Version

1.7.0.65

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10 pro

### Printer

Bambu Lab P1S

### How to reproduce

No go live option, no matter which printer you select.

### Actual results

No go live option, no matter which printer you select.

### Expected results

Go live, with being about to use OBS capture port.

### Project file & Debug log uploads

No crash, Just not an option for go live.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

rinytin, 2023-07-22:
Add the live view from the software is super laggy! Really can not even watch it. I run a mesh wifi, and my signals are very high.

mbaran5, 2023-07-23:
@rinytin The camera on the P1P and P1S is .5FPS. Yes, not 5FPS, but .5. 1/2 of one frame per second. You should not be seeing smooth motion.

bambu123, 2023-07-25:
> Add the live view from the software is super laggy! Really can not even watch it. I run a mesh wifi, and my signals are very high.

Go live is not supported for P1P and P1S because of low FPS.

closed 2023-07-25 by rinytin

4lve, 2023-11-29:
I would like the ability to atleast get an image from the printer

cstenkamp, 2024-11-05:
For a way to stream P1S video feed, see https://github.com/slynn1324/BambuP1Streamer