bambulab/BambuStudio #2101: Missing [orca] Callibration

VeryKross, 2023-07-22:
**Is your feature request related to a problem? Please describe.**
BambuStudio lacks helpful calibration features for the P1 series that were added [thank you!] in Orca. The latest BambuStudio build has merged in many of the Orca enhancements but has left out these very useful calibration features.

**Describe the solution you'd like**
I would really like to see Orca calibration features included in BambuStudio, perhaps further enhanced, to provide the P1 users the best tools possible to optimize our prints.

**Describe alternatives you've considered**
My only "work around" now is to run Orca to perform the calibrations, then copy those values back and apply them in Studio.

**Additional context**
[image]

zhimin-zeng-bambulab, 2023-07-25:
[image]
@VeryKross We have corresponding manual calibration for P1 printer.

VeryKross, 2023-07-25:
Thanks for the reply! I'm not seeing the rest of the calibration tests (Pressure Advance, Retraction Test, etc.).

zhimin-zeng-bambulab, 2023-07-26:
Flow Dynamics has the same effect as Pressure Advance, other types of calibration, we will gradually supplement in the future. Thanks for your proposal!

lanewei120, 2023-07-28:
> Thanks for the reply! I'm not seeing the rest of the calibration tests (Pressure Advance, Retraction Test, etc.).

@VeryKross 

hello

could you tell us which calibrations are you also using besides flow dynamics and flow rate?
and please help to give us the specific usage scenarios

thanks~

SnickCH, 2023-07-28:
@lanewei120 

Situation
- I have different brands of filament (from the same type like PLA, PETG etc.) that I print a lot and with AMS

Use-Cases
- When I print with AMS and color changes, I want all my filament to be as good as possible calibrated. Therefore I have to provide this values. The "auto-calibration" would only calibrate the first filament. 
- I print with 3rd Party plates (I don't like the bambulab ones ->glue stick....) and many of them do not really work well with the auto-calibration (and nether does the PEI from Bambulab). Therefore I need to provide my own values.

FYI: I use orca slicer, there I have a lot of tuning possibilities (as a reference)

lanewei120, 2023-07-28:
> @lanewei120
>
> Situation
>
> * I have different brands of filament (from the same type like PLA, PETG etc.) that I print a lot and with AMS
>
> Use-Cases
>
> * When I print with AMS and color changes, I want [...]

@SnickCH 
thanks for the detailed information

when you use orcaslicer, which calibrations are you using besides the PA and flow rate calibration?

SnickCH, 2023-07-28:
@lanewei120 
- PA + Flowrate (first)
- Tolerance (here you could probably use the x/y/z cube) to make sure, that it works for technical prints
Then depending on the profiles I have (only if necessary)
- Retraction

One thing that could be optimized (in orca) and you could keep in mind: the flowrate calculation is a bit pain (I do it in Excel), this could be done in the UI with just adding the "value" from the test and then the software calculates it.

FYI: I don't use bambu studio, therefore I don't know what it really provides etc.. I just saw this request and wanted to help with explaining what and why I'm using it in orca slicer

lanewei120, 2023-07-28:
> @lanewei120
> 
> * Temptower
> * PA + Flowrate
> * Tolerance (here you could probably use the x/y/z cube) to make sure, that it works for technical prints
> 
> Then depending on the profiles (only if necessary)
> 
> * Retraction
> 
> One thing that could be optimized (in orca): the flowrate calculation is a bit "complicated" (I do it in Excel), this could be done in the UI with just adding the "value" from the test and then the software calculates it.
> 
> FYI: I don't use bambu studio, therefore I don't know what it really provides etc.. I just saw this request and wanted to help with explaining what and why I'm using it in orca slicer

thanks for these informations

1. Are you using these calibrations for Bambu printer or for other printers?
2.  if using Bambu printer, is the temptower calibration still useful?

SnickCH, 2023-07-28:
@lanewei120 
I use orca slicer only for bambulab X1C

I use the temp tower for filament that I can't find a profile or the standard profile doesn't give me the results I want.

Example:
-  I found a perfectly working profile for eSun PLA (and eSun PLA printed perfectly with the generic and the "bambulab PLA" profile). 
- But prusament doesn't work for me, here I'm tuning with the temp tower to find out what temperature I should use, before I start the other tests.

skuridin, 2023-08-01:
I too find the "Temperature Tower" calibration very important, especially with PETG. Some brands require lower printing temperature than others.

nirurin, 2023-08-22:
Also the 'flow calibration' test being used in bambu studio is out of date, you need to use the newer 'Ellis' variant (Orcaslicer finally updated to include this version, and Studio should as well)

nirurin, 2025-06-13:
Temperature tower seems to be available in bambu studio... but only for non-bambu printers. 

Which is weird, because bambu printers need to do temp-tower calibrations for filaments just as much as every other printer. 

100% temp towers (and every other calibration option) needs to be added for every printer.