bambulab/BambuStudio #2103: Auto refill / filament backup not working (1.7 beta 2)

Waffull, 2023-07-23:
### Bambu Studio Version

1.7.0.65 (beta 2)

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

X1C combo

### How to reproduce

Method 1:

1. Load two spools with the same filament properties into the AMS.  In my case, two spools of Bambu ABS (one blue, one red - Same exact specs, only difference is color)
2. Synchronize filament from AMS
3. Go to Device tab
4. Click AMS settings (this icon should be changed to make it clear and should also display the name of the button when rolled over.  I get wanting to use symbols so you don't have to deal with different languages, but use universal symbols, at least.
5. Make sure AMS filament backup is checked
6. Hit Auto Refill (you really should be consistent with names... is this auto refill or filament backup.  Choose one, please.)
7. Error is displayed: 
> There are currently no identical spare consumables available, and automatic replenishment is currently not possible. (Note, replenishment would be replacing the spool with a new/different one, not simply switching the slot from which filament is fed.)

Method 2:

1. Resync filament from AMS
2. Hit Slice Plate, to slice the plate with your model on it.
3. Hit Print Plate
4. Hit the filament/slot shown to see which slots are loaded with the same material selected during preparing the model/plate.  In my case, I can see both Bambu ABS spools/slots showing, with the other two, which have PLA or ASA in them, not showing.)
5. Hit Auto Refill
6. Get error: 
> There are currently no identical spare consumables available, and automatic replenishment is currently not possible.

### Actual results

Error: Auto Refill
> There are currently no identical spare consumables available, and automatic replenishment is currently not possible.

Separately, there's also a bug where if you do manage to have two of the exact same spools, even if you have to lie to the software and tell it a different color from what it actually is, the printer prints with which ever of the two it already had loaded, even if your sliced model was done with a different slot.  This is very frustrating when you've set your colors and painted parts and the printer decides to ignore that and simply chooses what it prefers, as the first color.

### Expected results

Ability to specify which slot(s) and in what order, should be used if the primary slot used in the slicing runs out.

### Project file & Debug log uploads

[DW735_BlowerAdapter_Shopvac.zip](https://github.com/bambulab/BambuStudio/files/12138871/DW735_BlowerAdapter_Shopvac.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

DokuKaefer, 2023-07-23:
The problem is the different color.
Why should it continue to print with red if the main print job is blue?

MadMax389, 2023-07-23:
It would be nice to be able to assign any color (of the same material) when the spool runs out. Maybe an option to reload the current slot (as currently), or an option to select another slot (if you don't care about the color, or load the same color in the selected slot without setting it up in the AMS first). Even when you set everything up properly before printing, the slicer seems to take the first slot with the desired color, regardless of the slot selected. If my remnant roll is in a later slot than my full roll, I have to make sure and reassign the slot before printing. I've had several switchover failures because I forgot to assign the slot at print time.

[image]

Waffull, 2023-07-24:
Plenty of reasons why.

1. Most of us don't stock multiple rolls of the same colors, it's simply too expensive.
2. When printing functional parts, color doesn't really matter.  
3. Using up small amounts that are left on spools.
4. When a print requires more than a single spool of filament
5. etc, etc, etc.

But maybe most of all, because it makes sense to allow a print to finish when the properties of the materials are the same.  I get why the properties need to be the same, but it makes no sense to require the colors to be the same.  Bambu wants us to buy their filament but creates this limitation.  Until this is fixed, I personally won't be buying anymore Bambu filament because you can't override the color and therefore there is no way to do this.  One of the great things about having a system like an AMS should be the ability to just roll over to the next spool... who cares about color?  If one does, then simply don't set this functionality or spend the money and have a stock of the same colors.

Hologos, 2023-07-24:
Why don't you assign the same color to AMS slots if you want to use whatever color? What you propose would definitely confuse more beginners that would help experienced folks.

MadMax389, 2023-07-24:
> Why don't you assign the same color to AMS slots if you want to use whatever color? What you propose would definitely confuse more beginners that would help experienced folks.

Sometimes I forget to set up the AMS before printing and you can't reassign the color after you start. It would be nice to be able to select a spool when the filament runs out. It could be an addition to the existing notification dialog. "Reload empty spool or keep printing from spool X"

Waffull, 2023-07-25:
You can't change the color of Bambulab filament because it's determined
from the RFID, believe me, I have tried.

Regarding beginners, it's very simple, don't make it the default setting,
instead in the AMS settings window add an option to turn on using the same
material, regardless of color.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

SnickCH, 2023-08-26:
I would like to see the feature for the AMS
- Choose the order of the slots which should be used after each other (regardless of the color, as long as it is the same material): example order-> Slot 2, 4, 3, 1 (so first 2, then 4, then 3 and last 1)
- If it runs out and you don't have set the order of the slots, you should be able to choose which slot is used
- it should respect the temperature settings per slot that is set

dtdionne, 2023-09-03:
There are plenty of reasons indeed.  I don't care about color, I care about wasting filament.  As others have said, this is a no brainer Bambu and is used to work for me.  I have many two tone functional parts that work great and NO WASTED FILAMENT.

Now...NOW, I have wasted filament, time and electricity, and that pisses me off.

Get auto-refill working again Bambu, it's one of the main reasons I gave you my money for ur stuff.

ThorsReign, 2023-09-24:
Version 1.7 is now in release and this issue still exists on P1S with AMS.  AMS Slots 1 and 2 are both set to Generic PLA with the exact same color, clicking the auto-refill button gives the error as described above.  Exact same type and properties of filament.  This feature worked in 1.6.4.x prior to the release of 1.7 when it was a checkbox within the AMS settings rather than a standalone button.

GuillaumeLeclerc, 2023-09-27:
Same issue here, Slot 1 and 3 have exactly the same:
- Color
- Filament Type
- Profile on Bambu Studio

Forced a resync of the AMS in BambuStudio

Still get the message: "There are currently no identical spare consumables available [...]"

elmerohueso, 2023-10-01:
I'm struggling with this, trying to use up the last bits of one spool of red before starting another. Slots 2 and 4 are both Red (same RGB values) and Generic PLA set in the Device tab. I synced the Filament on the Prepare tab, set my model to use the filament type matching slot 4. When I send the print over, I tell it to use slot 4 (the one that's almost empty). Auto Refill says "There are currently no identical spare consumables available, and automatic replenishment is currently not possible."

Is this not the correct process to use "AMS Filament Backup"?

ThorsReign, 2023-10-01:
As I dig into it there seem to be two functionalities that do similar things, and one of them is broken.

AMS Filament Backup can be selected from the gear icon next to the AMS in the Device Tab of Bambu Studio - this automatically moves from one filament to the other if they have the same color/type/settings.  So if slots 2 and 4 are both red generic PLA and you're printing from slot 2 and it runs out it will switch to slot 4.

Auto-refill is supposed to do the same thing but you tell it what order to move through the slots.  It's documented so poorly I wonder if it's supposed to be for when you have multiple AMS connected, or at least 3 spools of the same type/color loaded.

MadMax389, 2023-10-01:
> As I dig into it there seem to be two functionalities that do similar things, and one of them is broken.
>
> AMS Filament Backup can be selected from the gear icon next to the AMS in the Device Tab of [...]

I believe AMS filament backup and auto refill are the same thing. Clicking auto refill checks to see if there are identical spools loaded for the filament backup option to work. This works well on my X1C. However, on my P1P, clicking auto refill indicates there are no identical filaments, even when there are. BUT, it does seem to work anyway, as the P1P actually did switch rolls when the remnant roll ran out, even though clicking auto refill failed to show the identical rolls.

elmerohueso, 2023-10-01:
> I believe AMS filament backup and auto refill are the same thing.
I'm inclined to believe that too, having seen screenshots of how it works with an X1C.

> BUT, it does seem to work anyway
I ran a job as a test last night and it did, indeed, automatically switch to the matching spool when the first one ran out.

So maybe this is just a visual bug with the slicer and/or P1P/P1S firmware, but I hope it gets fixed soon as it causes a bit of confusion.

Hologos, 2023-10-01:
> I'm struggling with this, trying to use up the last bits of one spool of red before starting another. Slots 2 and 4 are both Red (same RGB values) and Generic PLA set in the Device tab. I synced the [...]

I saw devs saying (couple of months ago) that the filament backup works only in ascending slot order. Seems pretty dumb but worth the try. Try selecting slot 2 and see if slot 4 will be selectable as backup.

elmerohueso, 2023-10-01:
> > I believe AMS filament backup and auto refill are the same thing.
> 
> I'm inclined to believe that too, having seen screenshots of how it works with an X1C.

My C++ is pretty weak, but my I took a look at the code.
[The "Auto Refill" (m_ams_backup_tip) label is shown if the selected machine meets various factors (including ams_auto_switch_filament_flag, which appears to be tied to the "AMS filament backup" checkbox).](https://github.com/bambulab/BambuStudio/blob/5ef759ce41863f989da7b363f7c94e5edc5ade0d/src/slic3r/GUI/SelectMachine.cpp#L2929)

> So maybe this is just a visual bug with the slicer and/or P1P/P1S firmware, but I hope it gets fixed soon as it causes a bit of confusion.

[Clicking "Auto Refill" opens [popup_filament_backup()](https://github.com/bambulab/BambuStudio/blob/5ef759ce41863f989da7b363f7c94e5edc5ade0d/src/slic3r/GUI/SelectMachine.cpp#L1131)
[popup_filament_backup() runs AmsReplaceMaterialDialog::update_machine_obj to populate the dialog](https://github.com/bambulab/BambuStudio/blob/5ef759ce41863f989da7b363f7c94e5edc5ade0d/src/slic3r/GUI/SelectMachine.cpp#L1510)
[The error message is being thrown by AmsReplaceMaterialDialog::update_machine_obj](https://github.com/bambulab/BambuStudio/blob/5ef759ce41863f989da7b363f7c94e5edc5ade0d/src/slic3r/GUI/AmsMappingPopup.cpp#L1335)

My guess is that there's a bug (or restriction) in [this block](https://github.com/bambulab/BambuStudio/blob/5ef759ce41863f989da7b363f7c94e5edc5ade0d/src/slic3r/GUI/AmsMappingPopup.cpp#L1304) in how the filaments are being compared to create "groups".

MadMax389, 2023-10-01:
> I saw devs saying (couple of months ago) that the filament backup works only in ascending slot order. Seems pretty dumb but worth the try. Try selecting slot 2 and see if slot 4 will be selectable as backup.

Actually the problem is that when you sync the same color/material in different slots to the slicer and assign the later slot as your remnant roll, the slicer will pick the first similar color to print. You have to reassign the correct slot at print time.

[image]

[image]

Hologos, 2023-10-01:
> Actually the problem is that when you sync the same color/material in different slots to the slicer and assign the later slot as your remnant roll, the slicer will pick the first similar color to [...]

I've already filed a bug 2 months ago #2190. It seems liie they break important features and don't fix them. :<

AnthonyDesigns-NL, 2023-12-31:
This problem is still here. 

[image]
A1 and A3 where/are both set to white ABS when clicking on the "Auto Refill" button it does show A3 as the next option:
[image]
It simply doesn't do anything, it's a feature that should simply work. It's advertised as something that's working but it doesn't.

miru8112, 2024-01-02:
P1S here. 
I think most is already said. 
I need this feature to save long prints from spools running out, bt also, I need this feature to use up small amounts of filament I have left over. I don't feel that having the same color should be a limitation, but that has also been said and discussed before. As long as its PLA, it should be a possible auto-fill option. Please make this happen... 

Currently I have rests of a few meters, too short to put in the AMS, but put on the external, there will also be no AMS to back em up when they run out. Its just the worst.