bambulab/BambuStudio #2104: PAINT: Maintain paint on Negative Part

TheAtomicSoul, 2023-07-24:
### Bambu Studio Version

1.7.0.65

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Printer

P1P with AMS

### How to reproduce

import a primative object and paint it two+ colors, turn it into a negative part type intersecting another model.

### Actual results

Negative part turns into a single primary filament color.

### Expected results

Negative part maintains painted on colors. 

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-07-25:
@TheAtomicSoul   In what scenario do you need to switch the object to a negative part and still want to keep its coloring information.

TheAtomicSoul, 2023-07-25:
> @TheAtomicSoul In what scenario do you need to switch the object to a negative part and still want to keep its coloring information.

I have models I purchased from a maker and want to add half a torus to it to make a loop I can thread through. 
On some models the area it passes through is composed of two different colors. 
As is, only the first layer of one side of the "tunnel" is the correct color, but everything after is the other sides color, even when the torus is painted with each side the appropriate color.

zhimin-zeng-bambulab, 2023-07-25:
@TheAtomicSoul   I didn’t quite understand what you meant. It might be better if you have a picture to explain your expected effect. What I understand is that if it is a negative part, the place covered by the negative part will be subtracted when slicing, and even if it has a color, it will not show up.

TheAtomicSoul, 2023-07-25:
The first layer is pink, but then the entire negative part is blue. 

> @TheAtomicSoul I didn’t quite understand what you meant. It might be better if you have a picture to explain your expected effect.

[image]
[image]
[image]