bambulab/BambuStudio #2106: "Slowdown for overhangs" speed overwrites the "Slow printing down for better layer cooling" option

Erwin-Repolust, 2023-07-24:
### Bambu Studio Version

1.7.0.65

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

P1P

### How to reproduce

In the filament settings on the "cooling" tab, activate the checkbox "Slow printing down for better layer cooling" and set a minimum layer time of 40 seconds and set "Min print speed" to 0.

In the global process settings in the "speed" tab make sure "Slowdown for overhangs" is checked and speed overrides are set to 100mm/s for every angle.

Now slice a basic object like a 3cm sphere.

### Actual results

The minimum layer time specified is overwritten by the 100mm/s. Layers will finish under 40 seconds.

### Expected results

The minimum layer time has priority over the speed specified in "Slowdown for overhangs". Layer time is 40 seconds for the whole object.

### Project file & Debug log uploads

[Sphere.zip](https://github.com/bambulab/BambuStudio/files/12153107/Sphere.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Erwin-Repolust, 2023-07-24:
[Sphere.zip](https://github.com/bambulab/BambuStudio/files/12153123/Sphere.zip)

Added the process setting to the project file.

Expected:
[image]

Actual:
[image]

QingZhangBambu, 2023-08-09:
I didn't get you. if the part is overhang, a slow speed guarantees the surface quality.

Erwin-Repolust, 2023-08-09:
As already stated. If I specify a minimum layer time that should have priority over the, in this case speedup that the overhang speed overwrite produces. I can not make it more clear.

QingZhangBambu, 2023-08-21:
why, these part may fail if the speed is not slowed down?

QingZhangBambu, 2023-08-21:
if you don't want it, you can unable the overhang option

GuzziRaz, 2023-09-15:
@QingZhangBambu The point is both settings should be honored. OP states that when slowing down for overhangs, the total layer time is erroneously ignored.

Erwin-Repolust, 2023-10-20:
@QingZhangBambu In the example I gave above instead of slowing the speed down for overhangs, the feature is actually speeding things up and thereby ignoring the minimum layer time specified elsewhere. Please reattach the bug label and let someone else read through it if u do not understand my writing.

Erwin-Repolust, 2023-10-20:
> if you don't want it, you can unable the overhang option

this is not a fix or a workaround! Layertime depends on layer surface area. There are models where the overhang speed would be a slowdown and on other models it would be a speed up in the current slicer so the issue persists.

QingZhangBambu, 2023-10-23:
Actually, it is a feature request. I've added it to our feature pool.

QingZhangBambu, 2023-10-23:
the concerned staff will evaluate it.

MrDulkes, 2024-01-13:
Just adding my opinion that this is a bug, not a feature, and should be fixed.

QingZhangBambu, 2024-03-05:
https://github.com/bambulab/BambuStudio/issues/3657

Erwin-Repolust, 2024-03-05:
> #3657

Even if this gets implemented which would be good - the bug above still needs to be fixed.
Its one check in the logic of the "Slowdown for overhangs" per layer.

After slicing with "Slowdown for overhangs" check if layertime is below minimum, if yes then disable the logic of "Slowdown for overhangs" for this layer. Easy stuff one would think.

QingZhangBambu, 2024-03-05:
[image]
@Erwin-Repolust this option could control if the overhang path be cooling

Erwin-Repolust, 2024-03-05:
@QingZhangBambu I know but I want both features to work simultaneously and not to turn one off cause it behaves erroneously.

QingZhangBambu, 2024-03-06:
@Erwin-Repolust  it works simultaneously, if the layer time is not satisfied, the overhang speed part will be reduced.

Erwin-Repolust, 2024-03-06:
@QingZhangBambu Just retested, you are right, it does work in the current version of bambu studio as expected

closed 2024-03-06 by Erwin-Repolust