bambulab/BambuStudio #2108: Adding Creality Cr-6 Se and Max printers Plus Jgmaker Artist D Pro for closable Profiles

bhast2, 2023-07-25:
**Is your feature request related to a problem? Please describe.**
My problem was that I had to switch between slicing software

**Describe the solution you'd like**
Have the printer profiles in the slicing software

**Describe alternatives you've considered**
using a different slicer soft ware

**Additional context**
The profiles are all ready made.
you can get them here just would like to see them added to the code.
Creality CR-6 Se and Max printers  https://github.com/bhast2/CR-6Se-Max-OrcaSlicer-profile/releases/tag/v1.5
JgMaker Artist D pro printer https://github.com/bhast2/JgMaker-Artist-D-Pro-OrcaSlicer-Profile/releases/tag/v1

SaltWei, 2023-07-26:
Thanks! We will have a check

bhast2, 2023-07-26:
awesome. thank you

bhast2, 2023-07-26:
i released v1.6.1 with a minor update. just a heads up

XunZhangBambu, 2023-07-27:
Thanks! We will select some typical nozzles

bhast2, 2023-07-27:
thank you very much I am working on adding a tpu profile

closed 2023-08-14 by XunZhangBambu