bambulab/BambuStudio #2109: Viewing the mesh in Paint mode

Kyrian-29, 2023-07-25:
[Feature request] When using "Paint" with the "Triangle" tool, have the possibility to visualise the mesh of the part to be coloured.
Trimming with the triangle tool is not easy without being able to clearly see the mesh (the triangles).
[image]
A view of the mesh like with NetFab for example would be easier to do a perfect trimming.
[image]

Haidiye00, 2023-07-27:
Okay, thank you for your suggestion. We will consider it soon

Haidiye01, 2023-12-01:
[image]
The latest version already has this feature, please experience it

lanewei120, 2023-12-02:
[image]
use a shortcur to turn it on/off

Kyrian-29, 2023-12-02:
TaaaaaDaaaaaaa !!!!!!
Oooohh Yessss !!!!!
Thank's

closed 2024-01-16 by Haidiye00