bambulab/BambuStudio #2110: Flow Dynamics not being applied for P1P

mbaran5, 2023-07-25:
### Bambu Studio Version

1.7.0.65

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

P1P

### How to reproduce

Run Manual Flow Dynamics Calibration 
Set Value 
Export 3mf of plate
View gcode

gcode indicates the following 
`enable_pressure_advance = 0`
`pressure_advance = 0.02`

note that pressure advance is off, and the advance value is not the one chosen during my calibration. 

### Actual results

Actual results are that print is running without pressure advance. 

### Expected results

Pressure advance should be set to 1 and the flow value should reflect the chosen value as a part of the manual process. 

When slicing with Orca and manually setting the PA value, I see the following configuration set in gcode

`enable_pressure_advance = 1`
`pressure_advance = 0.03`
`M900 K0.03 L1000 M10 ; Override pressure advance value`

### Project file & Debug log uploads

[dehyd.zip](https://github.com/bambulab/BambuStudio/files/12163743/dehyd.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

lanewei120, 2023-07-28:
hello @mbaran5 

for the dynamic flow calibration, we use a different solution as Orca

because the compensation parameter is not only relative to the viscosity of the melted filament , but also relative to the melt zone diameter , nozzle diameter , nozzle length . As there is tolerance in the manufacturing , the melt zone diameter , nozzle diameter , nozzle length will not perfectly match the design , so the final compensation parameter for an specific filament will be different on different machines , and for now there is no a good solution to decouple these factors.

so we stored the calibration values(K factor) on the printer side, not in the filament preset
we don't need to use M900 in gcode to apply this values

you can follow the wiki guide for the detailed steps on Bambu Studio

mbaran5, 2023-08-01:
Understood. So ensuring that the external slot is set with the correct K value and hitting print is all that is required to ensure that the K value is used? 

Are there plans to support storing a K value per filament preset now that we have the feature to manually check K on a P1P/S in the new software version?

zhimin-zeng-bambulab, 2023-08-03:
@mbaran5 
1. Yes, as long as the k value is set, the printer will use the k value you use. 
2. We will improve the P1P/S printers in the future.

cjccjj, 2024-02-25:
@zhimin-zeng-bambulab @lanewei120 

> @mbaran5
> 
> 1. Yes, as long as the k value is set, the printer will use the k value you use.
> 2. We will improve the P1P/S printers in the future.

How to set the K factor in the Device tab for Ext spool?  In the Prepare tab, there is filament selection, do you must select the filament with the Ext lable for the setting in Device tab to take effect?  the filament sections in Prepare tab and in Device tab are Not synced, what would happen if they don't match? 

Also, as K is store in machine, does it also stored in a gcode file that sent to printer? I mean if I print a job that I printed before, which is already in SD card, I start print from printer, without using Bambustudio to send a job, what K value would be used?

zhimin-zeng-bambulab, 2024-08-08:
The k value of each slot in the device page represents the k value used by each slot on this printer. 
It has nothing to do with what filament you use for slicing in prepare page.
This k value is not stored in the GCode file. If you start printing directly from the printer, the k value set on the slot will be used. If it has not been set, the default k value will be used.