bambulab/BambuStudio #2111: Project Section (Tab) not editable and not showing information

JDPS, 2023-07-26:
### Bambu Studio Version

1.6.24/ 1.7.0.65

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU: Intel Core i5 9600K
MB: Z390-A Pro
Memory: 48GB DDR4-3000
GPU: NVIDIA GeForce GTX 1660 SUPER

### Printer

Bambu Lab X1 Carbon Combo

### How to reproduce

TestA:

1. Open Bambu Studio;
2. Click on "New Project /n Create new Project";
3. Click on the "Project" tab;

TestB:

1. Open Bambu Studio;
2. Open any saved project file (i.e. .3mf file) - Click on "File"  and select "Open Project";
3. Select a saved project file from the local hard drive;
4. Click on the "Project" tab;

### Actual results

TEST_A:

1. Display of a empty box and with the message "no model information"

TEST_B:

1. Display on the left column of three tabs: "Model Information", "Accessories", "Profile Information";
2. The last two tabs are not selectable;
3. No information is editable;

### Expected results

TEST_A:

1. A vertical column on the left which includes the following tabs: "Basic Info", "Pictures", "Bill of Materials", "Assembly Guide", "Others" and in the right are the editable fields of each one of the aforementioned tabs. E.g. for the "Basic Info", it includes  "Author" and "Model Name".

TEST_B:

1. The project section includes a vertical column on the left which includes the following tabs: "Basic Info", "Pictures", "Bill of Materials", "Assembly Guide", "Others" and in the right are the editable fields of each one of the aforementioned tabs. E.g. for the "Basic Info", it includes  "Author" and "Model Name". The editable fields may already contain information from previous save.

### Project file & Debug log uploads

[projectfile.zip](https://github.com/bambulab/BambuStudio/files/12168938/projectfile.zip)
[debug_network_Wed_Jul_26_07_41_03.log](https://github.com/bambulab/BambuStudio/files/12168941/debug_network_Wed_Jul_26_07_41_03.log)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

LJKraus, 2023-07-26:
This feature really needs to be restored.  It is very useful to record notes about a project.

Or remove the tab altogether.

lanewei120, 2023-07-27:
hello, we are re-implementing this feature
and will open it in recent versions

JDPS, 2023-07-28:
Thank you. 
It is good to know that this feature is within the plan.

mbukosky, 2023-08-08:
can't wait to see it

JDPS, 2023-09-22:
@lanewei120 Lane Wei, 
A gentle reminder that many are eager to get this feature back (e.g. [Bambu Studio - What happened to the project section?](https://forum.bambulab.com/t/bambu-studio-what-happened-to-the-project-section/16159)
I hope that the BL team may find time to implement it. 
TY.

YifanWuBambu, 2023-10-10:
Hi @JDPS ,
The project tab in bambu studio is now read only.
It displays information of the models downloaded from MakerWorld.
[image]
[image]
[image]

lanewei120, 2023-10-24:
close this issue as the solution has been published

closed 2023-10-24 by lanewei120

dmmop, 2023-12-29:
That is a workarround, not a solution. 

And is terrible for a WIP when you need edit that information multiple times.

LJKraus, 2023-12-29:
It's not a workaround, it is a fundamental change in the purpose and use of the tab that destroys user control over the privacy and security of their work.  Forcing every project into the cloud is ridiculous.  The correct solution would be to restore the original capabilities of the tab, allowing the data to be stored as a local .3MF file, with optional uploading to MakerWorld.

The closest thing to a solution I can find is in OrcaSlicer.  Add your project information (text only) to the "Notes" section of the process profile, then save the profile as a "Preset within Project" before saving the project .3MF locally.

t0nyz0, 2023-12-31:
Yea, I still have projects all the time that are completely incorrect naming.. I go to the project tab, non-editable.. very very frustrating. All my cloud project history have incorrect naming for the project. This is not a fix, you guys broke it worse.

t0nyz0, 2023-12-31:
Why is this closed?

LJKraus, 2023-12-31:
It's closed because it is working exactly the way Bambu wants it to work.  

If we don't like it, don't use it.

t0nyz0, 2023-12-31:
"Dont like it dont use it" is a cop out. You cannot modify project info on the project tab, all cloud prints that are locally created projects have the incorrect name when sent to the cloud. All of the print history that is stored in the cloud is incorrectly named... its a pretty obvious bug.  @LJKraus

bambu123, 2024-01-02:
> 为什么这个关闭了？这个实施是错误的，这张票的期望没有得到满足。开发人员提出了一个为最终用户提供很少甚至没有功能的解决方案……然后就关闭了票证。这是不好的做法。@JDPS您应该在没有任何评论的情况下重新打开此问题，因为未提供解决方案。另外，黑客（不是解决方案）是强制用户通过makersworld 网站来打印具有正确名称的项目？！？！？我听得对吗？这是我从 Bambu 开发人员那里看到的最糟糕的问题“修复”之一。没有开发出解决方案，这纯粹是懒惰。
> 
> @bambu123 @walterwongbbl @StoneLiBambulab 有没有办法重新打开此问题并获得实际修复？

This feature has been migrated to the model store.
This feature has been migrated to the model store. You can upload and define models here.
https://makerworld.com/en

t0nyz0, 2024-01-03:
@bambu123 We understand how it works now. You should **not** be required to have a makerworld account to edit a project name. The project tab should allow you to edit the project name without uploading the model to makerworld. 

The reason for being able to have a correct project name is so that the print history on the app accurately represents what the person is printing. I have "Christmas Ornament" in my print history for STL files I've loaded that weren't related to the "Christmas Ornament" makersworld project. There are bugs in the way that the projects are named, sometimes projects inherit old makerworld project names that are no longer loaded in BambuStudio. If we had a way to verify the project name before printing we wouldnt have incorrect project history. The Project tab should allow that functionality as requested in this ticket.

jbrewlet, 2024-02-01:
Still needing this feature. I'm still baffled that when I send a project to my X1 from Bambu Studio and set the name it does nothing and just reverts to whatever is brought into the "Model Name" under project.

Why can't I name the file what I want?

Its an organizational nightmare.

pacmanmt, 2024-02-15:
@BambuLab By now it should be clear that your customers (me included) don't want to forcibly move their intellectual work on your cloud. There have been tons of doubts around your policy of forcing people to work on the cloud, you should just place back the old project functionality, period. 
Furthermore the past issues of printers starting to print by themselves because a mess (or it was an hack?) on your cloud, should have clearly pointed you on the right direction: leave the users to freely decide to put their work on the cloud without withdrawing useful functionalities.

bambu123, 2024-02-18:
> @bambulab By now it should be clear that your customers (me included) don't want to forcibly move their intellectual work on your cloud. There have been tons of doubts around your policy of forcing [...]

This function may need to be re-planned. What content do you need to edit so much?

pacmanmt, 2024-02-18:
> This function may need to be re-planned. What content do you need to edit so much?

Just the whole Project tab.
This functionality is so cool because it can help us to better organize the projects, documenting them with pictures and notes.
I personally have hundreds of projects, some dating back some years ago; having them well documented would help a lot.
This feature together with the multi-plate functionality are top-notch.
I don't see what could be the issue of restoring the functionality that was there with the previous versions of Bambu Studio, you don't even have to write the code from scratch!

howsTricks, 2024-05-13:
Hi, what is the status of this issue? I have many files that have the wrong model info.

SaltWei, 2024-05-13:
> Hi, what is the status of this issue? I have many files that have the wrong model info I cannot edit anything on this tab.

Will be improved in next version

howsTricks, 2024-05-14:
Thanks for the update, looking forward to it!

jinhei, 2024-07-16:
> Will be improved in next version

Can we re-open this issue then? Or is this tracked somewhere else that we can subscribe to?

t0nyz0, 2024-07-26:
Still dealing with this and no changes in the new version.

I printed something from MakerWorld and when it finished I cleared the STL out out of BambuStudio. 
I then imported some other STLs in and started another project only to realize hours later that the Project tab is still tied to a the Makerworld print. If the project tab is locked to a Project then bambustudio should not allow you to import other STLs. This is a disaster of a bug and it seems that Bambu just doesnt care. I wasted at least 2 hours just on this one instance. Since you cant clear what project its locked to you have to literally start all over again with a fresh BambuStudio instance. 
Bambu has likely cost me dozens of hours just because they dont want to fix this issue. 

Also if you just double click on an STL on your desktop it will open into the active BambuStudio window, which could be a makersworld project. From that point on that 3MF/STL will forever be locked to that project. A very simple fix is just to have every STL open in a new window, especially if the current window is locked to a makersworld project.  You should not be able to double click and it open into the current project, it should start a new one. If you want an STL to be added to your current project you should be required to use the import option. 

I think at this point Bambu either does not care about this issue or simply does not understand the problem and how to fix it. But either way, this is one bug that is so irritating that I'm tempted to never open another MakerWorld project again.

Jamyn, 2024-12-15:
> This function may need to be re-planned. What content do you need to edit so much?

- Sometimes I **don't** want to upload my design, I'm learning or it's just for myself. I can't do it without now without uploading.
- Sometimes I want to add links to accessory item sources, or price information (ex: screw sizes and stores)
- Sometimes I want to add notes about any problems I have with this particular model (do not change setting X or Y happens)

littlejackal, 2025-08-05:
Setting aside the very real issue of whether it is or is not desirable to use Makerworld as an intermediary, this change has effectively removed my ability to iterate a project until it is printable before I can make notes on it. 

Makerworld has strict requirements for projects, and for me to make notes on any of my in-flight projects I would need to make sure they're printable, and I can provide photographic evidence of that. Until then, my only option is to keep a text file in the directory next to my project file in hopes that I don't forget notes I made in iteration. 

If the intention is to make Makerworld the source of truth for project information, for consistency or completeness, etc., then simply make the user aware that once they upload it to Makerworld any notes they made will be imported into or subsequently overwritten by the project information FROM Makerworld. 

Whether or not a less generous interpretation of the motivation behind this change exists, the fact is that I can't keep in progress notes on the project, in the section which any reasonable end-user would expect to be able to keep information about a project in. As a result I have to keep sidecar notes outside of the application.