bambulab/BambuStudio #2115: Tree support bug?

shawnishihara, 2023-07-27:
### Bambu Studio Version

1.7.0.65

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 13

### Printer

Bambu Lab X1C

### How to reproduce

[image]

Support settings

### Actual results

[image]

Floating supports inside opening of model. (view from bottom)

### Expected results

Not have floating supports. This does not show up with "normal" support

[image]

### Project file & Debug log uploads

Have not printed since installing the public beta 1.7.0.65

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tangketan, 2023-08-02:
Please upload the 3mf file for diagnosis.

shawnishihara, 2023-08-02:
[MalteseFalcon.3mf.zip](https://github.com/bambulab/BambuStudio/files/12243789/MalteseFalcon.3mf.zip)

.3mf zipped and uploaded. 

I'm seeing the same bug on different models, not just this one.

UndeadKernel, 2023-08-05:
I'd like this issue to be solved. I'm therefore also attaching a .3mf file that exhibits problems with floating supports.
This is specially obvious starting around layer 101 (near the legs of the miniature).

[floating_supports.3mf.zip](https://github.com/bambulab/BambuStudio/files/12268125/floating_supports.3mf.zip)

gbibgim, 2023-08-06:
I can confirm this too, weird floating tree supports. Sometimes it gets stuck while generating them on 65%. BS 1.7.2.51
[image]
[entrence_reducer_mount.zip](https://github.com/bambulab/BambuStudio/files/12268948/entrence_reducer_mount.zip)

igiannakas, 2023-08-07:
Same here. Pretty annoying issue rendering tree supports as useless.

jaytigner, 2023-08-08:
Same problem. 
[image]
[image]

jaytigner, 2023-08-08:
The new hot fix did not solve the issue. These pictures are after downloading the newest release. 
[image]
[image]
[image]
[image]

tangketan, 2023-08-10:
Yes, this is a bug with Linux and Mac on which I didn't test enough. Fix should be released soon.

shawnishihara, 2023-08-10:
[image: Screenshot 2023-08-09 at 10.41.04 PM.png]

Still a major; work-stop bug after hotfix 1.7.3.50

UndeadKernel, 2023-08-10:
I see support problems also under Windows. They do seem a bit less "extreme" but there are some issues still with supports and support interfaces appearing out of nowhere.

[image]

tangketan, 2023-08-11:
Thanks for reporting the bug. We've found the problem. The solution is on the way.

SaltWei, 2023-08-16:
Will update new version including fix for this issue soon！

gbibgim, 2023-08-16:
[image]
Looks good now!

SaltWei, 2023-08-17:
Included in latest release https://github.com/bambulab/BambuStudio/releases/tag/v01.07.04.52.

closed 2023-08-17 by SaltWei

sockmonkey0223, 2023-10-14:
I'm using v01.07.07.88 and I still see this issue.