bambulab/BambuStudio #2116: [External spoolholder filament settings available in the Prepare tab] External spoolholder filament settings available in the Prepare tab

BambuLabDiscord, 2023-07-27:
Make the External spoolholder filament settings available in the Prepare tab if you want to print from the external spoolholder. Right now you have to change one of the spools in the AMS to whatever filament is on your externals spool and remember to switch of the AMS when it comes to printing. Telling the Prepare tab to not use the AMS and letting you set the external spools filament settings for slicing would be much easier and will stop me from trying to print PLA with ABS settings like I did because I forgot to toggle off "Use AMS"...

zviratko, 2023-08-02:
This is actually not true, since some time ago you see the filament on the external spool as a "fifth" (or whatever number you have in AMS+1) filament in the slicer and you can select it for the object.

outrun0506, 2023-12-16:
@zviratko Can you please explain where this option should be? If a sync my AMS i see only the loaded Spools, there is no fifth option. I always act as described in the first Post.