bambulab/BambuStudio #2118: "Only one wall on top surfaces" feature is only applying to topmost surface

cutmoney, 2023-07-27:
### Bambu Studio Version

1.7 beta 2

### Where is the application from?

Bambu Lab github releases

### OS version

MacOS 13.4.1

### Additional system information

MacBook Pro 2021 M1 Max

### Printer

Bambu Lab P1

### How to reproduce

1. Turn on "Only one wall on top surfaces" (which is enabled by default already)

### Actual results

One wall only applies to the uppermost feature of a model.

### Expected results

It should apply to the flat surfaces of all top layers, as it is a "top surface" for all other logic (line width, speed, etc.). This is how it used to function in Bambu Studio 1.6.2.

### Project file & Debug log uploads

[Test.3mf.zip](https://github.com/bambulab/BambuStudio/files/12188779/Test.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

T00LF00L, 2023-07-28:
This is the correct behavior.  It only applies to the uppermost feature, exactly as described.

cutmoney, 2023-07-28:
> This is the correct behavior. It only applies to the uppermost feature, exactly as described.

That's not how it previously worked, nor how any other "top" parameter works. If I choose a top surface pattern, it applies to all top surfaces, not just the uppermost one. If I choose "top surface" line width or speed, it applies to all top surfaces, not just the uppermost feature.

That would be inconsistent with how it works in the release version and how every other slicer works.

QingZhangBambu, 2023-08-08:
In Arachne mode, we implement the "only one wall top" function and split it into three modes, you can get that in hotfix2 soon.

closed 2023-08-09 by QingZhangBambu