bambulab/BambuStudio #2124: Walls on top surface don't adhere to speed for top surface

Hologos, 2023-07-31:
### Bambu Studio Version

1.7.0.65

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 13.5 (22G74)

### Printer

BambuLab P1P

### How to reproduce

1. Open included 3mf.
2. Go to Preview.
3. Select speed view.

### Actual results

Only solid infill is printed with speed for top surface, walls are printed faster (even faster than on lower layers). See included 3mf file.

[image]

### Expected results

The whole top surface is printed at speed for top surface.

### Project file & Debug log uploads

[top-surface-speed.3mf.zip](https://github.com/bambulab/BambuStudio/files/12215254/top-surface-speed.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2023-09-05:
wall has wall speed, if you want to change it, you can use the modifier to set it.

QingZhangBambu, 2023-09-06:
what do you need? if you only want to get better print quality of small text, adding a modifier to reduce it would be fine. Or you just want the all-top path to use top surface speed?

Hologos, 2023-09-06:
I thought that top speed is meant for the whole top surface. The odd thing is that the wall is printed with lower speed the whole model except for the topmost layer. It seemed like a bug to me. If it's inteded, I'll try to use the modifier.

QingZhangBambu, 2023-10-24:
Top surface speed means top infill surface. it may be confusing. I will talk to the concerned staff about it.

closed 2024-01-18 by SaltWei