bambulab/BambuStudio #2126: Move away from Print when Layer time is less than X

Festivejelly, 2023-07-31:
**Is your feature request related to a problem? Please describe.**
Due to the high speeds at which the X1C prints when layer times differ significantly in time this can cause banding to appear, with some spots looking much more shiny than others.

**Describe the solution you'd like**
I would like a parameter I can control on a per filament basis so that if the layer time is less than X then the toolhead will move away from the print and it will wait until that time has passed. Before resuming it will go over to the purge bucket and prime the nozzle before moving over to the prime tower to get rid of any oozing and to prime the nozzle ready to print the next layer on the print.

**Describe alternatives you've considered**
Slowing down the print introduces more severe banding as printing fast creates a matte appearance for PLA and PLA+ (200mm/s) slowing down to say 100mm/s causes the print to show shiny bands, as such this is undesirable.
Slowing down the infill significant can mitigate this issue but its a very manual process to slow down the infill time on the sections that print too quickly, as you don't want to slow the infill down for the whole print.

**Additional context**
Additional screenshots and photos of prints will be provided soon.

QingZhangBambu, 2023-08-01:
You can unable that to not slow down the print speed. Does that what you wanted?
[image]

Festivejelly, 2023-08-01:
> You can unable that to not slow down the print speed. Does that what you wanted? [image]

Slowing the print has undesirable effects. When a print is slowed down to aid cooling it causes more banding, the slower parts become much more shiny.

I want an option were I can keep the printing speed as it is, but allow the layer to cool before printing the next layer.

My workaround at the moment is to print a sacrificial primitive and set the print speed for that primitive as very slow.

Its not ideal, a better option would be to dwell over the purge bucket, purge, then prime on the prime tower.

Xiuqi-Huang-CN, 2023-08-02:
@Festivejelly To axoid the prints get different looks of normal and shinier (the lower-speed areas look shinier), we recommend that users set the entire outer wall speed to be close to or the same as the lowest-speed areas, and this will works better than your advise. We have wiki for this kind of cases: https://wiki.bambulab.com/en/knowledge-sharing/common-print-quality-problem
[image]
Hope that you try our method and tell us if it work for your prints.

Festivejelly, 2023-08-02:
Thanks for the suggestion but this still causes quality issues. Even if I print at say 50mm/s on the outer wall for all layers. Layers that take significantly longer still show banding because the layer had more time to cool. Slowing the print time down so significantly is undesirable as it leaves a shiny quality to it. Where as I prefer the matte appearance.

This is what it looks like with my workaround of printing a sacrificial part that I can vary the speed on
[image]

In contrast to this you can see there is a big difference in layer times:
[image]

Simply slowing the whole print down isnt really a desirable solution as the whole print then becomes shiny and the X1C seems to produce more surface artifacts when its printing slowly.

So thats why I want an option: Move head away from print if layer time is less than X, then on resume, purge, prime and print next layer.
It might not be suitable for everyone, if there was a parameter for layer time Id write a post processing script to do it, but there isnt.

QingZhangBambu, 2023-08-03:
@Festivejelly how about inserting M400 S cmd to layer change gcode to pause the extrude to filament tower manually?

QingZhangBambu, 2023-08-03:
@Festivejelly  And pause extrude may cause oozing which may influence surface quality.

Festivejelly, 2023-08-10:
> @Festivejelly And pause extruder may cause oozing which may influence surface quality.

This is why I suggest between the pausing we could use the purge bucket and then prime.

This could be an option, the downside is we waste material, but for a high quality print its worth it.

Festivejelly, 2023-08-10:
> @Festivejelly how about inserting M400 S cmd to layer change gcode to pause the extruder manually?

We have no way to determine how long to pause for. We need to be able to understand how long each layer takes so the operation would be:

If layer time < desired minimum layer time
 then move to purge bucket
 wait until (desired minimum layer time - layer time)
    purge nozzle
move to prime tower
prime
resume print

I understand this process would waste a lot of filament potentially but that's why id like this feature added as an option, so people can turn this on or off, maybe under the experimental section.

I print highly detailed parts and I'm willing to sacrifice a bit more material for quality, but I understand not everyone shares this viewpoint.

Maybe we don't need to purge part just the nozzle prime.

Maybe instead the flow could be:

If layer time < desired minimum layer time
 then retrace layer with ironing settings

SaltWei, 2023-08-18:
I think maybe we can develop one experimental options.
User can choose to slow down or moving away when layer time is not enough.

sapper4484, 2023-09-14:
plus one on seeing this issue. I am seeing the same issue on using Bambu PLA Silver and would like a fix.

SrPhilippe, 2024-01-18:
> plus one on seeing this issue. I am seeing the same issue on using Bambu PLA Silver and would like a fix. Not sure if the pictures help. ![20230914_224418](https://private-user-images.githubusercontent.com/94809329/268119407-3d7829f8-aea7-4514-b9f5-abfb025b943a.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU2MTI3NDYsIm5iZiI6MTcwNTYxMjQ0NiwicGF0aCI6Ii85NDgwOTMyOS8yNjgxMTk0MDctM2Q3ODI5ZjgtYWVhNy00NTE0LWI5ZjUtYWJmYjAyNWI5NDNhLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE4VDIxMTQwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTcwYjNmY2ViNTM3NmE0MDY4NjQwZjA5MzU3NDBkNGMxMjgxMTY1YTkyYWIxZTc5YmFhZDcxZDI3N2MxYjQxNGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.7XijW91stodaHPtyNdbDiXq6_OnqyHgDqX5Ic8NozEk) ![20230914_224414](https://private-user-images.githubusercontent.com/94809329/268119429-33252d37-8925-46c2-b7d9-1ceb44564ab9.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU2MTI3NDYsIm5iZiI6MTcwNTYxMjQ0NiwicGF0aCI6Ii85NDgwOTMyOS8yNjgxMTk0MjktMzMyNTJkMzctODkyNS00NmMyLWI3ZDktMWNlYjQ0NTY0YWI5LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE4VDIxMTQwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBmYzc0YjA4YjMzMTEwZjFkZTUzZTBkYzFjNTI2ZTgyODc0MjIxNjRmNWFmYTEzYTAzZTcwZDAxNmEwMDNhN2MmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.TE3V3p0HtmHEDQTQVpH_XYKGEk2UCP9OJ7eleHHw9H4) ![20230914_221631](https://private-user-images.githubusercontent.com/94809329/268119455-5faba888-1263-42c0-a2f4-e289d8b33837.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU2MTI3NDYsIm5iZiI6MTcwNTYxMjQ0NiwicGF0aCI6Ii85NDgwOTMyOS8yNjgxMTk0NTUtNWZhYmE4ODgtMTI2My00MmMwLWEyZjQtZTI4OWQ4YjMzODM3LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE4VDIxMTQwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFhOGI5OGVkMjliZGQ1OTY4ZTU1OGM0YTZiMGU4MTRkOGY3MzM1OGQ0MGNjNTkzMmY1MzgzMzU4OTBjMDNhMGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.iqRKd9gpFmb0SgAuUA81NVYoMcLOWQtYXAP7qiVdq7M) ![20230914_224126](https://private-user-images.githubusercontent.com/94809329/268119532-8497c2c3-b90c-4611-a178-62086b141886.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU2MTI3NDYsIm5iZiI6MTcwNTYxMjQ0NiwicGF0aCI6Ii85NDgwOTMyOS8yNjgxMTk1MzItODQ5N2MyYzMtYjkwYy00NjExLWExNzgtNjIwODZiMTQxODg2LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE4VDIxMTQwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY5ZjA3M2EzNTE5YmZlMGI3OTExZmFmMjEzMjA3Y2NhNjVhZDM4ZTBlNWE5MTRmZThmNWVjNWJiY2I2NTUwYjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.euSUwQpr_fRoo8loDrI2a_hp-5CtGTZbFZzPzkE7jmY)

Did u fixed it?

SrPhilippe, 2024-01-18:
@sapper4484 Did u fix it?

kvnper, 2024-02-29:
I proposed a solution here:

https://github.com/bambulab/BambuStudio/issues/3657

DrFischie, 2024-10-06:
They need to add a feature to let the nozzle wait until a specific layertime is met. This will make the whole OW look the same. Cura had a feature like this but it also needs to step away from the model so there won't be any filament that ozzles out of the nozzle and lands on the print. You could also force it to print a primetower to make layertimes longer in general but it adds more waste to the print, which isn't ideal.

Option in cura:
[image]

Here are some pictures so visualize the problem. Increasing the layertime under cooling only reduces the print speed overall and creates a difference in speed. That difference in speed will create a similar issue, so this isn't a valid option.

Speed:
[image]

Flow:
[image]

Layertime:
[image]

Printed model at 0,12 mm and 50 mm/s OW:
[image]