bambulab/BambuStudio #2128: High System Load

twalenzyk, 2023-07-31:
### Bambu Studio Version

v1.7.0 Public Beta 2

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10 (22H2)

### Additional system information

CPU: Intel(R) Core(TM) i7-4800MQ CPU @ 2.70GHz   2.69 GHz^
Memory: 16GB
Display Card: Intel HD Graphics 4600

### Printer

Bambu Lab P1P

### How to reproduce

1. Install 1.7.0 Pub Beta 2
2. Print something
3. View live video 

### Actual results

Live video stalls after some time and finally results in black image.
System Load is very high, fan is running full speed.

### Expected results

Live video works uninterrupted.
System load stays normal

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12222793/log.zip)
[Unbenannt.zip](https://github.com/bambulab/BambuStudio/files/12222807/Unbenannt.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2023-08-01:
@twalenzyk What about the old 1.6 version?

twalenzyk, 2023-08-01:
I switched back to 1.6.2.4 an it does not show the same behaviour.

bambu123, 2023-08-01:
Are you sure it is relating to the live view? Please give me a snapshot video with cpu performance.

bambu123, 2023-08-24:
@twalenzyk Any news?

bambu123, 2023-08-28:
Close it first, and reopen it if there is a new example.

closed 2023-08-28 by bambu123