bambulab/BambuStudio #2130: [FEATURE REQUEST] A switch to be able to (temporarily) sort the RECENTLY OPENED file list alphabetically.

bolsoncerrado, 2023-08-01:
**Is your feature request related to a problem? Please describe.**
Given you added the setting to preferences where a user can choose how many recently opened files to display, when list starts to grow, it may be hard to locate a file. While it can still be done via the  OS "Finder"/"File Explorer", may be just a SWITCH button to temporarily sort the files displayed alphabetically could help too without having to leave Bambustudio.

**Describe the solution you'd like**
A button next to CLEAR ALL to sort files either ALPHABETICALLY or by MOST RECENT. For example: if pressed: alphabetically, if unpressed: by date, as usual.
Optional: ascending/desecending sort order button.

RyanEwen, 2023-08-04:
I would love for the home screen of the application to have sorting options as well, and some settings as to how many projects to remember.

bolsoncerrado, 2023-08-04:
If you mean how many RECENT projects to remember, this was already implemented and you can choose it on settings/preferences.

bolsoncerrado, 2024-05-07:
This feature request is long due too @lanewei120 @bambu123 :)