bambulab/BambuStudio #2132: BambuStudio 1.7 beta 2 segfault on Arch Linux

ajacocks, 2023-08-01:
### Bambu Studio Version

01.07.01.56

### Where is the application from?

Built from source by yourself or others

### OS version

Arch Linux

### Additional system information

```
Memory:
  System RAM: available: 7.76 GiB used: 572.6 MiB (7.2%)
  Array-1: capacity: 8 GiB slots: 1 EC: Multi-bit ECC
  Device-1: DIMM 0 type: RAM size: 8 GiB speed: N/A
CPU:
  Info: 8-core model: Common KVM bits: 64 type: MCP cache: L2: 32 MiB
  Speed (MHz): avg: 1992 min/max: N/A cores: 1: 1992 2: 1992 3: 1992 4: 1992
    5: 1992 6: 1992 7: 1992 8: 1992
Graphics:
  Device-1: driver: bochs-drm v: N/A
  Display: server: X.org v: 1.21.1.8 with: Xwayland v: 23.1.2 driver: X:
    loaded: modesetting gpu: bochs-drm tty: 80x24 resolution: 1280x800
  API: OpenGL Message: GL data unavailable in console and glxinfo missing.
Drives:
  Local Storage: total: 100 GiB used: 75.81 GiB (75.8%)
  ID-1: /dev/sda vendor: QEMU model: HARDDISK size: 100 GiB

$ gcc --version
gcc (GCC) 13.1.1 20230429
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

### Printer

X1 Carbon w/ AMS

### How to reproduce

1. clone from git
2. build using BuildLinux.sh (with minor mods for Arch)
3. cd build/package && ./bambu-studio

Mods to BuildLinux.sh:
```
$ diff ./BuildLinux.sh-orig ./BuildLinux.sh
7,8d6
< FOUND_GTK2=$(dpkg -l libgtk* | grep gtk2)
< FOUND_GTK3=$(dpkg -l libgtk* | grep gtk-3)
85,86c83
< if [ ! -d "build" ]
< then
---
> if [ ! -d "build" ]; then
91,95c88,102
< export REQUIRED_DEV_PACKAGES="libmspack-dev libgstreamerd-3-dev libsecret-1-dev libwebkit2gtk-4.0-dev libosmesa6-dev libssl-dev libcurl4-openssl-dev eglexternalplatform-dev libudev-dev libdbus-1-dev extra-cmake-modules"
< # libwebkit2gtk-4.1-dev ??
< export DEV_PACKAGES_COUNT=$(echo ${REQUIRED_DEV_PACKAGES} | wc -w)
< if [ $(dpkg --get-selections | grep -E "$(echo ${REQUIRED_DEV_PACKAGES} | tr ' ' '|')" | wc -l) -lt ${DEV_PACKAGES_COUNT} ]; then
<     sudo apt install -y ${REQUIRED_DEV_PACKAGES} git cmake wget file
---
> if [[ $( lsb_release -sd | grep -E ^Ubuntu >/dev/null 2>&1 ; echo $? ) == 0 ]]; then
>   # ubuntu
>   export REQUIRED_DEV_PACKAGES="libmspack-dev libgstreamerd-3-dev libsecret-1-dev libwebkit2gtk-4.0-dev libosmesa6-dev libssl-dev libcurl4-openssl-dev eglexternalplatform-dev libudev-dev libdbus-1-dev extra-cmake-modules"
[... 139 lines cut ...]
openat(AT_FDCWD, "/usr/share/fonts/TTF/SegUIVar.ttf", O_RDONLY) = 39
fcntl(39, F_SETFD, FD_CLOEXEC)          = 0
newfstatat(39, "", {st_mode=S_IFREG|0644, st_size=1851000, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 1851000, PROT_READ, MAP_PRIVATE, 39, 0) = 0x7fb4a2e3c000
close(39)                               = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
rt_sigaction(SIGSEGV, {sa_handler=SIG_DFL, sa_mask=~[RTMIN RT_1], sa_flags=SA_RESTORER, sa_restorer=0x7fb69d24fab0}, NULL, 8) = 0
rt_sigreturn({mask=[]})                 = 94045903182128
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
+++ killed by SIGSEGV (core dumped) +++
```

### Expected results

application runs

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12233454/log.zip)
[debug_Tue_Aug_01_15_50_34_127401.log.0.zip](https://github.com/bambulab/BambuStudio/files/12233478/debug_Tue_Aug_01_15_50_34_127401.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

ajacocks, 2023-08-01:
I confirmed the same problem with the pre-built Ubuntu appimages from 1.7 beta 2 and v01.06.02.04.

foosinn, 2023-08-01:
Had the same issue. Fedora images seem to work? o.O

lanewei120, 2023-08-02:
hello @ajacocks 

is it possible to use gdb to run the application and catch a core dump stack,
then we can see the crash stack

ajacocks, 2023-08-03:
Sure. I rebuilt with debug set, and captured a core dump, attached.
[core.95586.bz2.gz](https://github.com/bambulab/BambuStudio/files/12246984/core.95586.bz2.gz)

lanewei120, 2023-08-05:
> Sure. I rebuilt with debug set, and captured a core dump, attached. [core.95586.bz2.gz](https://github.com/bambulab/BambuStudio/files/12246984/core.95586.bz2.gz)

I am afraid I don't know how to use such file
could you show the call stack directly? or tell us how to use your file for debugging?

lanewei120, 2023-09-13:
thanks @fang64 

this stack seems strange
did you build this by using the newest codes on master branch?

and when this crash happens, is the main window already shown?

fang64, 2023-09-16:
> thanks @fang64
> 
> this stack seems strange did you build this by using the newest codes on master branch?
 This is using the tagged release not master.
> 
> and when this crash happens, is the main window already shown?

Main Window does show, but it crashes after logging in.

boredomwontgetus, 2023-10-23:
Exactly the same issue here on Arch. All AppImages segfault before any window. Own builds also segfault immediately.

ajacocks, 2023-10-23:
Confirmed here, as well, on Arch.

- Alex

lanewei120, 2023-10-30:
@ajacocks 

which arch linux version are you using?

we are considering to setup an arch linux machine

fang64, 2023-11-01:
@boredomwontgetus and @ajacocks  you should move ~/.config/BambuStudio somewhere else, and then try to run it. I had that issue I found out due to the network plugin, felt pretty dumb.

boredomwontgetus, 2023-11-01:
I already did. I ensured a clean $XDG_CONFIG_HOME with every version and package i tried. I figured out Bambu_Studio_linux_fedora_v01.06.02.04-20230427094209.AppImage is the latest version running somehow.

lanewei120, 2023-11-09:
Hello 

we have just released v1.8.0 public beta version
and we have fixed some crash issues on linux,
also we have verified it on an Arch Linux computer, it works fine here

could your guys have a try also?
https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.57

boredomwontgetus, 2023-11-09:
Hi!

I tried both AppImages (ubuntu, fedora) and the result is the same as before.
Maybe its worth mentioning i use Wayland with a tiling Compositor named "Hyprland".

lanewei120, 2023-11-10:
> Hi!
> 
> I tried both AppImages (ubuntu, fedora) and the result is the same as before. Maybe its worth mentioning i use Wayland with a tiling Compositor named "Hyprland".

@boredomwontgetus 

thanks for the fast reply
I can try to change the compositor to "Hyprland" for a try

By the way, do you know how to change it?

fang64, 2023-11-10:
@boredomwontgetus that's interesting, I am using Wayland on Gnome. I'll poke hyprland as well and see if the fedora appimage blows up as well. that's wild. I'm also curious if you have libwebkit2gtk installed as well?

sebasortiz-dev, 2023-11-10:
I have the same issue, with Manjaro (Arch Linux) with X.org (no wayland) but with i3 as window manager. The libwebkit2gtk is installed 

```
Downloads/Bambu_Studio_linux_ubuntu_v01.08.00.57-20231109141031.AppImage
[2023-11-09 19:59:12.681048] [0x00007f211b86e480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
loading network module at /home/sebas/.config/BambuStudio/plugins/libbambu_networking.so
error, dlerror is /home/sebas/.config/BambuStudio/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)

(bambu-studio:162804): GLib-GObject-CRITICAL **: 19:59:14.296: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.296: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:162804): GLib-GObject-CRITICAL **: 19:59:14.296: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.296: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:162804): GLib-GObject-CRITICAL **: 19:59:14.297: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.297: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:162804): GLib-GObject-CRITICAL **: 19:59:14.298: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.298: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:162804): GLib-GObject-CRITICAL **: 19:59:14.301: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.301: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:162804): GLib-GObject-CRITICAL **: 19:59:14.302: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.302: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.358: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.361: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.361: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.362: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.362: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.362: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.362: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.362: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.362: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.366: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.367: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.367: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.440: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.440: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.447: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.447: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.450: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.450: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.454: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.454: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.458: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.458: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.462: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.462: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.466: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:162804): Gtk-CRITICAL **: 19:59:14.466: gtk_widget_set_size_request: assertion 'width >= -1' failed
Segmentation fault (core dumped)
```

boredomwontgetus, 2023-11-10:
> By the way, do you know how to change it?

you just install wayland and hyprland via pacman. login to one of your ttys and execute hyprland.

boredomwontgetus, 2023-11-10:
> That dependency is not included in the appimage for whatever reason, I suspect it's just how the appimage build script is setup, but that's one of the culprits for why they had to split fedora/ubuntu since that was creating grief and segfaults.

Thats an interesting piece of information. I do have it installed though:
```
❯ yay -Q | grep webkit
webkit2gtk 2.42.1-2
webkit2gtk-4.1 2.42.1-2
```

fang64, 2023-11-18:
@boredomwontgetus 
The Ubuntu version expects webkit2gtk 4.0-37 
The Fedora version wont segfault with  newer verisons, I have webkit2gtk-4.1 2.42.2-1

lanewei120, 2024-08-08:
similiar as https://github.com/bambulab/BambuStudio/issues/3973
close this issue here

closed 2024-08-08 by lanewei120