bambulab/BambuStudio #2133: Web requests to Amazon, Apple and Microsoft

foosinn, 2023-08-01:
### Bambu Studio Version

1.7.0 beta 2

### Where is the application from?

Bambu Lab github releases

### OS version

Linux

### Additional system information

--

### Printer

p1p

### How to reproduce

* download App Image
* run in terminal
* see log messages

### Actual results

there are a ton of web requests to Apple, Amazon and Microsoft to the `/` URL.

Is this some sort of online check?

It also seems to contain information about the printer:

X-BBL-Client-Name: BambuStudio
X-BBL-Client-Type: slicer
X-BBL-Client-Version: 01.07.00.65
X-BBL-Device-ID: *removed*
X-BBL-Language: en-US
X-BBL-OS-Type: linux
X-BBL-OS-Version: 6.4.6

### Expected results

Not connecting to the internet

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2023-08-02:
it is a bug in the app, you should report that to Bambu handy. The concerned staffs will check that.

foosinn, 2023-08-02:
This happend to me in the desktop application. Isn't this the correct place to report it?

QingZhangBambu, 2023-08-02:
@foosinn Sorry, I misunderstood. I thought app means application in your phone.

AtomicRPM, 2023-08-05:
Looks like the code has been removed from v01.07.02.51.  Good!

lanewei120, 2023-08-05:
already delted in v1.7.2.51
thanks for finding these

closed 2023-08-05 by lanewei120