bambulab/BambuStudio #2135: Failed to Install The Plug-In

RuiMarquesCardona, 2023-08-02:
### Bambu Studio Version

1.7.0.65

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

P1P

### How to reproduce

Fresh install of beta software over old main version, automatically let it try to install the network plugin.

### Actual results

The plugin will not install, will ask to check if blocked or deleted fby antivirus which it isn't

### Expected results

Install the plugin

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12234814/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

RuiMarquesCardona, 2023-08-02:
Possible Solution: Complete unistall of both Bambu Studio, and Orca Slicer if exists, and delete appdata folder.

Absolucy, 2023-08-02:
I've found that for some reason, Nvidia Broadcast had one of the DLL files in the `%AppData%/BambuStudio/plugins` folders opened?

brunopiovan, 2023-08-02:
fix for me was to just just deleted the %AppData%/BambuStudio/plugins folder

dagnarf, 2023-08-02:
"Intel ARC Control service" on my Samsung Galaxy Book 360 / Intel Xe graphics in Windows 11 Services has the same .dll file open and won't let it be deleted or overwritten - I stopped the service from Windows Services, was able to delete the files (and download the plugin)

jammydodger06, 2023-08-02:
> I've found that for some reason, Nvidia Broadcast had one of the DLL files in the `%AppData%/BambuStudio/plugins` folders opened?

Thank you!!!

StoneLiBambulab, 2023-09-11:
please look at this wiki page:
https://wiki.bambulab.com/en/software/bambu-studio/failed-to-get-network-plugin

Thanks!

Stone Li

StoneLiBambulab, 2023-10-24:
close this issue. open a new one if there is any problem.

Thanks!

Stone Li

closed 2023-10-24 by StoneLiBambulab