bambulab/BambuStudio #2137: Failed to install network plugin

seler, 2023-08-02:
### Bambu Studio Version

v01.07.01.62

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

P1S

### How to reproduce

1. Have existing working Bambu Studio installation
2. Install Bambu Studio v1.7.1
3. Start Bambu Studio
4. Observe the error

### Actual results

Got the error "Fail to install plug-in".
[image]

### Expected results

Network plugin should either be already installed or should install automatically without problems like in previous versions.

### Project file & Debug log uploads

[debug_network_Tue_Aug_01_15_22_42.log](https://github.com/bambulab/BambuStudio/files/12237708/debug_network_Tue_Aug_01_15_22_42.log)
[debug_Wed_Aug_02_10_03_50_18388.log](https://github.com/bambulab/BambuStudio/files/12237717/debug_Wed_Aug_02_10_03_50_18388.log)
[debug_Wed_Aug_02_10_03_18_1768.log](https://github.com/bambulab/BambuStudio/files/12237718/debug_Wed_Aug_02_10_03_18_1768.log)
[debug_Wed_Aug_02_10_02_50_20748.log](https://github.com/bambulab/BambuStudio/files/12237719/debug_Wed_Aug_02_10_02_50_20748.log)
[debug_Tue_Aug_01_15_22_42_22768.log](https://github.com/bambulab/BambuStudio/files/12237720/debug_Tue_Aug_01_15_22_42_22768.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Candyffm, 2023-08-02:
This is a duplicate of https://github.com/bambulab/BambuStudio/issues/2135
Try closing Nvidia Broadcast as it accesses a dll file inside the plugin folder (as suggested in the other issue). Worked for me.

seler, 2023-08-02:
Closing NVIDIA broadcast bypasses the problem but I still think it should be addressed. Closing this one in favor of #2135.

closed 2023-08-02 by seler