bambulab/BambuStudio #2138: [Feature Request] - Show auto-calibration values (from X1C)

SnickCH, 2023-08-02:
**Is your feature request related to a problem? Please describe.**
For some build plates (textured pei) the auto-calibration doesn't work. Or if you print with AMS multiple spools (different colours/vendors etc.) with different settings in the same print, the calibration only works for the first filament (if the build plate is supported .

**Describe the solution you'd like**
I would like to have a way in Bambu Studio to get the values from the auto-calibration. Or even better: +A way to safe it to a specific filament profile. 

**Describe alternatives you've considered**
I have to manually calibrate each filament with orca-slicer.

lanewei120, 2023-08-03:
@SnickCH 
hello, we have added this logic in the newest version of BambuStudio(1.7.1) and the X1C

could you update both the studio and X1C version for a try?

SnickCH, 2023-08-03:
@lanewei120 Thank you very much for the information. I updated the Firmware and Bambu studio.

Where do I find this information in the bambu studio?

zhimin-zeng-bambulab, 2023-08-03:
@SnickCH 
You can refer to this wiki, which introduces the use of calibration and the query of results.
wiki link: https://wiki.bambulab.com/en/software/bambu-studio/calibration_pa

SnickCH, 2023-08-03:
@zhimin-zeng-bambulab Thank you for the link and the well done explanation in the wiki. I will test this today.

closed 2023-08-03 by SnickCH