bambulab/BambuStudio #2139: [Feature Request] - add more nozzle wipes (Manually)

SnickCH, 2023-08-02:
**Is your feature request related to a problem? Please describe.**
Printing with PETG makes a lot of filament stick to the nozzle.

**Describe the solution you'd like**
I would like to have the option to
- A) have a button to manually wipe the nozzle, if the initial nozzle wipe was not enough
- B) Add (on a layer height base) the option for additional nozzle wipes, that will be directly done during the print. 

**Describe alternatives you've considered**
Pause the print, clean it, resume

pichen2021, 2023-08-04:
Thank you for your good idea, we will discuss this feature in future firmware.

Panda631, 2023-08-04:
In principle, this is a good idea. Even with other FIlaments, residues often remain on the nozzle.

RyanEwen, 2023-08-04:
Periodically automatically wiping the nozzle after a certain amount of filament is extruded, or certain # of layers have printed, would be super nice, especially for PETG.

This would further set Bambu printers apart from the competition that still don't see the value in the nozzle wiper.

Also think this could be a worthwhile upgrade to the wiper: https://www.printables.com/model/494670-bambu-lab-x1cp1p-nozzle-wiper-with-2mm-silicone-sh

hummelm10, 2023-12-31:
When/if this feature is added can there also be an option to wipe more often during purges? Instead of purging 600 do purge 200, wipe, purge 200, wipe, purge 200, wipe. I've noticed that sometimes the purges are quite large and piling up to the nozzle.

nietaki, 2024-10-03:
I don't really understand the sequence that happens before the print starts, but after the lines on the sides are printed (flow dynamics calibration?) but before the "actual" print starts, the nozzle usually has a glob of material on it (Bambu PETG, Bambu X1C + AMS) - and all of the nozzle wipes happened a long time ago.

Doing one last wipe before the print starts could be helpful, especially when printing with a brim - there's no risk of missing material, because the flow is going to stabilize long before the brim is finished (at least for bigger brims and larger parts, which is my use-case 99% of the time)

SnickCH, 2024-10-04:
> Periodically automatically wiping the nozzle after a certain amount of filament is extruded, or certain # of layers have printed, would be super nice, especially for PETG.
>
> This would further set [...]

I have this upgrade you linked and since then I have a much better result. Combine this with the "more wipes" and we are good to go :)

mnott, 2025-02-10:
This may help:

1. I removed the Extrusion Line at the back to protect that part of the print plate
2. I use a tape to protect the calibration area of the print plate
3. I added a simple cube of 0.1mm height to provide me  with some additional nozzle wipe, in a random place around where the extrusion line was

I'll illustrate with pictures.

1. I have removed the extra extrusion lines at the back right of the print plate, as because they are always printed in the same place, they'll be very hard to remove, leading to a damaged print plate pretty soon.

For this part, there are how2s out there; basically, my machine at the end, after the `T1000` line (around line 636) looks like this:

```
T1000
G1 E{-retraction_length[initial_no_support_extruder]} F1800
G1 X128.0 Y253.0 Z0.2 F24000.0;Move to start position
G1 E{retraction_length[initial_no_support_extruder]} F1800
M109 S{nozzle_temperature_initial_layer[initial_no_support_extruder]}
;G0 X253 E6.4 F{outer_wall_volumetric_speed/(0.3*0.6)    * 60}
;G0 Y128 E6.4
;G0 X252.5
;G0 Y252.5 E6.4
;G0 X128 E6.4
```

In other words, I commented out those lines that show the extrusion line. On a more recent firmware, they've put in a comment

```
;===== purge line to wipe the nozzle ============================
```

and I assume everything after that you can comment out.

2. I use a tape to protect the calibration zone; should I really need to print into that area, I can quickly remove that tape after the calibration:

[image]

3. Now, because of that change, the nozzle will not be very clean at the start of the print, and hence I ended up in this thread, I simpliy added a cube of 0.1mm height to the back right area (near where the extrusion line was, but in a more random place so it does not wear out the print plate):

[image]

[image]

So in other words, the place of that extra cube matters. If I put it to the front left, it will be printed last. At the back right, it'll be printed first.

Hope, this helps