bambulab/BambuStudio #2141: Flow Rate calibration auto mode output is always 0,00

Obicom, 2023-08-02:
### Bambu Studio Version

1.7.1

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

1. Go to "Calibration"
2. Choose "Flow Rate"
3. Choose " Auto Mode"
4. Wait for finishing the process

### Actual results

I did several tests but the outcome is always 0,00 and you can't save this value.

[image]

### Expected results

I would expect a serious value around ~1,00

### Project file & Debug log uploads

[empty.txt](https://github.com/bambulab/BambuStudio/files/12241513/empty.txt)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Sterbo88, 2023-08-02:
I too have the same problem… I tried various materials and various colors

cryd-s, 2023-08-02:
same problem here -.-

Datt-Flo, 2023-08-02:
same here 
flow rate calibration 0,000

zhimin-zeng-bambulab, 2023-08-03:
@Obicom  Could you provide the log file when you encounter an error situation?

Obicom, 2023-08-03:
> @Obicom Could you provide the log file when you encounter an error situation?

There is no real error situation ... @zhimin-zeng-bambulab 
The calibration of flow rate delivers an wrong value after the test without an error message.

zhimin-zeng-bambulab, 2023-08-04:
@Obicom 
Could you share your log files, when the calibration of flow rate delivers an wrong value after the test. The log file location is as follows image:
[image]

zhimin-zeng-bambulab, 2023-08-04:
@Sterbo88 @cryd-s @Datt-Flo  Could you also help provide the log file when the result is incorrect?

Sterbo88, 2023-08-04:
> @Sterbo88 @cryd-s @Datt-Flo Could you also help provide the log file when the result is incorrect?

here is the log

cryd-s, 2023-08-04:
[debug_Fri_Aug_04_09_05_58_3328.log.zip](https://github.com/bambulab/BambuStudio/files/12258219/debug_Fri_Aug_04_09_05_58_3328.log.zip)

Datt-Flo, 2023-08-04:
[image]
[debug_Fri_Aug_04_09_11_13_2160.log.0.zip](https://github.com/bambulab/BambuStudio/files/12258283/debug_Fri_Aug_04_09_11_13_2160.log.0.zip)

zhimin-zeng-bambulab, 2023-08-04:
Thank you very much for your cooperation! In this case, the printer calibration result is wrong, you can choose to recalibrate, or use manual calibration. We will improve the accuracy of the printer’s automatic calibration later.

cryd-s, 2023-08-04:
Thanks for your answer - but how can it be that it works for some and not for us. Is there a hardware defect here? 

Am just a little disappointed that there is no solution here.

Obicom, 2023-08-04:
@zhimin-zeng-bambulab it would be great if you could explain why this is only happening for "some" users but not for all. What is the root cause of this behavior?

Datt-Flo, 2023-08-04:
> Thank you very much for your cooperation! In this case, the printer calibration result is wrong, you can choose to recalibrate, or use manual calibration. We will improve the accuracy of the printer’s automatic calibration later.

Of course, we are happy to help find the cause of the error, since it is also in our interest to get a remedy as soon as possible.
To their answer; how often should I repeat the test in their opinion? At some point, my supply of filament is then also exhausted.
As an end user, I would be very grateful if a solution is quickly shown there.

Übersetzt mit DeepL (https://www.deepl.com/app/?utm_source=ios&utm_medium=app&utm_campaign=share-translation

dragon-eyes-bbl, 2023-08-04:
Hi All，
Thanks for your patience. We are still investigating the bugs at the moment and your feedback is very helpful. 
The flow rate calibration was a new feature, and we are still improving its compatibility and accuracy. 
Before the new firmware arrives, you may try the solutions below. 
- Clean Lidar camera len. 
- Check any strings or debris on the calibration patterns.
- Check if the filament is compatible. Silk, transparent, sparkles etc are incompatible. 

This article gives you more info about this calibration.  https://wiki.bambulab.com/en/knowledge-sharing/flowrate-calibration-by-microlidar

Obicom, 2023-08-04:
@zhimin-zeng-bambulab 

- Clean Lidar camera len. -> I did it already after first try without success
- Check any strings or debris on the calibration patterns. -> printed patterns looking good an clean
- Check if the filament is compatible. Silk, transparent, sparkles etc are incompatible. -> I tried it with PLA green and black (no silk or sparkles)

cryd-s, 2023-08-04:
I do all advises but everytime i got 0.000 

Tested several colors no silk sparkles etc.

dragon-eyes-bbl, 2023-08-04:
Hi @cryd-s Could you upload your machine's log to our customer service and provide me the ticket number? Thanks

cryd-s, 2023-08-04:
Of course please Tell me how to geht the log

lanewei120, 2023-08-04:
https://wiki.bambulab.com/en/x1/troubleshooting/export-failed-print-log-to-sd-card
https://wiki.bambulab.com/en/p1/troubleshooting/how-to-upload-log

> Of course please Tell me how to geht the log

cryd-s, 2023-08-04:
Ticket No: US230802492001

Datt-Flo, 2023-08-05:
After installing the new Bambu Studio 1.7.2.51 and the new firmware 01.06.01.00, the following error now occurs 

[image]

Sterbo88, 2023-08-05:
> Dopo aver installato il nuovo Bambu Studio 1.7.2.51 e il nuovo firmware 01.06.01.00, ora si verifica il seguente errore
> 
> [image]

it had already come out to me once… among the thousand tests…

Obicom, 2023-08-05:
Same here ... no longer zero value but the common error message ... it is not realy an improvement ;)

[image]

cryd-s, 2023-08-05:
@dragon-eyes-bbl did you see anything in the logs? I hope for a quick fix

Merlwynd, 2023-08-05:
Also, less important but…spelling and grammar. Believe last sentence should read “The failed test result will be dropped”…or better yet “….will be discarded”

zhimin-zeng-bambulab, 2023-08-07:
If the result of the automatic mode is incorrect, you could use the manual mode to calibrate instead. 
We will fix the problem of automatic mode as soon as possible.

dragon-eyes-bbl, 2023-08-07:
> @dragon-eyes-bbl did you see anything in the logs? I hope for a quick fix

Hi @cryd-s , sorry for the late reply. I saw your log and as this function is heavily linked to firmware, the improvement will be on with the firmware release.

cryd-s, 2023-08-07:
> > @dragon-eyes-bbl did you see anything in the logs? I hope for a quick fix
> 
> Hi @cryd-s , sorry for the late reply. I saw your log and as this function is heavily linked to firmware, the improvement will be on with the firmware release.

Thank you for your reply. I hope the claim to Bambu Lab is to bring a fix very quickly to avoid the loyal community and bad news.
Thank you for your efforts - do you have a rough idea when such a fix can be released.

dragon-eyes-bbl, 2023-08-11:
@cryd-s @Obicom I have locate the problem. It was the numbering format cannot be parsed in the message to printer for locale in some languages.  Bug fixing will be in the next version. 
At the moment, please try to change the language to English and retry the calibration.
Thanks for your patience.

Sterbo88, 2023-08-11:
> language in bamboo studio?

dragon-eyes-bbl, 2023-08-11:
> > language in bamboo studio?

Yes, the language in bambu studio

Sterbo88, 2023-08-11:
> @cryd-s @ObicomHo individuato il problema per il valore che restituisce 0.000. Era il formato di numerazione che non può essere analizzato nel messaggio alla stampante per le impostazioni locali in alcune lingue. La correzione dei bug sarà nella prossima versione.
> Al momento, prova a cambiare la lingua in inglese e riprova la calibrazione. 
> Grazie per la vostra pazienza.

> @cryd-s @Obicom I have locate the problem for value returning 0,000. It was the numbering format cannot be parsed in the message to printer for locale in some languages. Bug fixing will be in the next version.
> At the moment, please try to change the language to English and retry the calibration.
> Thanks for your patience.
language in bamboo studio?

Sterbo88, 2023-08-11:
> > > language in bamboo studio?
> 
> Yes, the language in bambu studio

thanks, as soon as I get back from the mountain (tomorrow) I'll try

Obicom, 2023-08-11:
I can confirm, after changing to English, Flow Rate is now working. Hopefully it will be fixed soon in a release version.

[image]

dragon-eyes-bbl, 2023-08-12:
> @dragon-eyes-bbl
> 
> So what does it mean, if I change the slicer from German to English, it should work ?

Yes, that is the current walk around for this version. 
We will fix that in the next release.

SaltWei, 2023-08-17:
Included in latest release https://github.com/bambulab/BambuStudio/releases/tag/v01.07.04.52.

closed 2023-08-17 by SaltWei