bambulab/BambuStudio #2142: Not responsiove and need to forcequit when returning from sleep since last beta (MacOS)

dragosmiron, 2023-08-02:
### Bambu Studio Version

1.7.1.62

### Where is the application from?

Bambu Lab github releases

### OS version

macos 13.4

### Additional system information

M2 Macbook Air

### Printer

P1P

### How to reproduce

Have Bambu Studio Opened
Close the lid of the Mac
Open it after a while
Application is not responsive and needs Force Quit

### Actual results

Needs to Force Quit the app

### Expected results

App should work

### Project file & Debug log uploads

[Archive.zip](https://github.com/bambulab/BambuStudio/files/12242139/Archive.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

dragosmiron, 2023-08-04:
[log.log](https://github.com/bambulab/BambuStudio/files/12265020/log.log)

dragosmiron, 2023-08-15:
[crash.log](https://github.com/bambulab/BambuStudio/files/12343221/crash.log)

walterwongbbl, 2023-08-24:
@dragosmiron 
Thank you very much. We will fix this issue

3dprinthobi, 2023-08-25:
Hi, exactly same issue happening with my MacBook Pro, 
i have to force quit the app to use it again.

walterwongbbl, 2023-10-13:
@3dprinthobi @dragosmiron 
Please download and use the latest version of Studio

closed 2023-12-18 by walterwongbbl