bambulab/BambuStudio #2144: Automatic Flow Calibration way to high K-Value

ThoughtfulDev, 2023-08-02:
### Bambu Studio Version

1.7.1.62

### Where is the application from?

Bambu Lab github releases

### OS version

macOS

### Printer

Bambu Lab X1C

### How to reproduce

1. Do automatic Flow Calibration
2. Calibrate K-Factor automatically with Cold-Plate and Glue

### Actual results

In two cases the K-Factor was way to high

Jayo Black PLA: 0.039
Extrudr Red PLA NX2: 0.042

My manual calibration using Ellis Pattern in OrcaSlicer Main branch (https://github.com/SoftFever/OrcaSlicer/pull/1547) gave me those values:
Jayo Black PLA: 0.02
Extrudr Red PLA NX2: 0.016

Printing a bambu v1 cube with the automatic calibration values results in gabs on top layer (because too high k-value) and "wobbly" corners (see image)

Pictures of Automatic calibration with k-value of 0.039 (you can see the wobbly corners and gaps on top)
[image]

[image]

Pictures of manual calibration with value of 0.02
[image]

[image]

### Expected results

Automatic K-Factor should be somewhere near manual calibration.

### Project file & Debug log uploads

None

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SaltWei, 2023-08-03:
Printer team need log and calibration pattern photos. Maybe the extrusion head is crooked, the external parameter is wrong, and finally the calculation line is wrong.

Could you move to https://forum.bambulab.com/ or BanbuHandy APP to report ticket (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log) including the printer log and calibration pattern photos? Thanks!

SaltWei, 2023-08-03:
By the way, after uploading the log and pictures, could you share the ticket number so that we can ping the printer team to check asap ? Thanks!

igiannakas, 2023-08-03:
Same issue here. I value is too high by 40% or so as well.also have over extrusion with the flow rate calibration using the LiDAR (as in the other issue I opened)

sanguinaria-canadensis, 2023-08-04:
this is a bug that the compensation model in the Studio and the MC (machine controler)  is not the same , so the calibration result Studio sent is not compatible with the MC .
We are now reverting the model used in the MC to the same as Studio to make things work , the new firmware will be ready in a few hours.
T_T.....

igiannakas, 2023-08-04:
> this is a bug that the compensation model in the Studio and the MC (machine controler) is not the same , so the calibration result Studio sent is not compatible with the MC . We are now reverting [...]

Fantastic thank you!!!

sanguinaria-canadensis, 2023-08-04:
> > this is a bug that the compensation model in the Studio and the MC (machine controler) is not the same , so the calibration result Studio sent is not compatible with the MC . We are now reverting the model used in the MC to the same as Studio to make things work , the new firmware will be ready in a few hours. T_T.....
> 
> Fantastic thank you!!! Is this also related to #2152 ?

well , #2152 is an other story , the other guy is looking at it ...

igiannakas, 2023-08-04:
this issue appears fixed in the latest release. k values from the lidar match the manual calibration.

ThoughtfulDev, 2023-08-05:
I agree. Seems fixed.

closed 2023-08-05 by ThoughtfulDev