bambulab/BambuStudio #2145: The buttons for feeding and returning materials cannot be displayed

szifan, 2023-08-02:
### Bambu Studio Version

01.07.07.62

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

windows 11

### Additional system information

12600kf，rtx3080，32g 6400hz，

### Printer

P1P

### How to reproduce

After updating to the latest version, only one retry button will be displayed on the device page

### Actual results

Unable to feed through the client, the mobile app can operate normally
[image]

### Expected results

Normal display of all buttons

### Project file & Debug log uploads

nothing

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-08-03:
@szifan 
It looks like you're printing?

szifan, 2023-08-03:
> @szifan
> It looks like you're printing?

No

walterwongbbl, 2023-08-03:
@szifan 
Do you only have one printer？

walterwongbbl, 2023-08-03:
Hi. @szifan 
We will add entrances for load filament/unload filament. Currently, you can use an extruder or use your app for operation.
[image]

lanewei120, 2023-08-05:
this issue should be fixed on the newest version v1.7.2.51
please have a try

closed 2023-08-05 by lanewei120