bambulab/BambuStudio #2148: New calibration flow test ignores "flushing volumes".

TheGadge, 2023-08-02:
### Bambu Studio Version

1.7.1.62

### Where is the application from?

Bambu Lab Official website

### OS version

Win11 PRO

### Additional system information

AMD Ryzen 9 5950X 16-Core Processor         
32 GB RAM
Geforce RTX 2060 6GB

### Printer

Bambu Lab x1c

### How to reproduce

"Calibration" top centre.
Auto calibrate *with two colours or more from AMS*

Heavy bleed from previous colour into next,
so:
Change flushing volumes to 2x factor.. (560 currently, after trying 1.5)
-No change.

### Actual results

With red first filament, and clear in bay 2, my 2nd test comes out pink with red still in it diluted, so it's not 100% the bay2 filament. No matter what I set in flushings.

### Expected results

flushing to be tied to global flushing settings, not it's own flush value for the test,
Tried software and machine reboot, even rebooted PC, no change, flush values ignored.

### Project file & Debug log uploads

n/a - built-in test that I can't change the parameters.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-08-03:
@TheGadge Not flushing it clean will not affect the final calibration result, so it doesn’t matter.

TheGadge, 2023-08-03:
> @TheGadge Not flushing it clean will not affect the final calibration result, so it doesn’t matter.

So the 2nd filament with 20-50% of a totally different filament from the first wouldn’t change how the test results came out?

can’t the flush just be linked to the global setting in slicer? Thanks for response.

zhimin-zeng-bambulab, 2023-08-04:
> So the 2nd filament with 20-50% of a totally different filament from the first wouldn’t change how the test results came out?
>
> can’t the flush just be linked to the global setting in slicer? Thanks [...]
Each time you change the material, you have purged the necessary amount of purge, and the remaining part, the prime line will purge again, and it will be clean when you really calibrate, even if there is still some residue, the impact on the overall is very small.

TheGadge, 2023-08-04:
Many thanks! 👌

closed 2023-08-07 by zhimin-zeng-bambulab