bambulab/BambuStudio #2150: Camera freezes regulary since update

Kawu42, 2023-08-02:
### Bambu Studio Version

1.7.1.62

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

P1P 01.03.00.00

### How to reproduce

(Not LAN-Mode) Open Bambu Studio, start a print, go on device, start camera stream, wait. After some minutes (1-5 usually) the stream just freezes. I have to stop the stream and start again to make it work again.

On the 01.06.02.04 before, it worked just fine. Got all of the 0,5 FPS I needed. :D

### Actual results

Camera Stream freezes.

### Expected results

Camera Stream does not freeze.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12244056/log.zip)
[Gridfinity Baseplate 2x5 Skadis.zip](https://github.com/bambulab/BambuStudio/files/12244061/Gridfinity.Baseplate.2x5.Skadis.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2023-08-03:
Same AS #2091

bambu123, 2023-08-03:
@Kawu42 
I found it may relate to the old P1P firmware, please wait for the 01.04 version of P1P update.

Kawu42, 2023-08-04:
> @Kawu42 But your logs are all from the 1.6 version, why?
> 
> such as: 2023-08-02 19:33:59.593723[Thread 0x0000142c]:last app version {1.6.2}, current version {1.6.2}

To be honest I can't tell you why the last print in the logs was on the 1.6 instead of 1.7.
Is it possible that the log gets filled when you close bambu studio? Because when I look into the logs, now the 1.7 is displayed. I will upload the updated logs in a minute above.

But that will be just for completion. I will just wait for the 1.04 for the P1P. Thank you for now.

Kawu42, 2023-08-07:
> @Kawu42 I found it may relate to the old P1P firmware, please wait for the 01.04 version of P1P update.

Todays P1P 01.04. Update made that the stream in bambu studio 1.7. does not freeze anymore, as you've predicted.

closed 2023-08-07 by Kawu42