bambulab/BambuStudio #2151: Could not resume after first layer "error"

quazzie, 2023-08-02:
### Bambu Studio Version

1.7.1.62

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

X1C

### How to reproduce

1. Print file
2. Get "error" "first layer problem, resume if acceptable"
3. Get no resume
4. Must cancel print, no resume on printer or bambu studio.

### Actual results

No resume

### Expected results

Resume

### Project file & Debug log uploads

Don't have. Moved on, didn't think to save them.
Later figured out i could submit bug report.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SaltWei, 2023-08-03:
@sodachen22 Please have a check on printer side

Bloodybooger, 2023-08-03:
I just experienced this same error. had to cancel print because there is no resume button

sodachen22, 2023-08-04:
There is a firmware bug for resuming from the first layer error. It will be fixed in the next version