bambulab/BambuStudio #2152: Automatic Flow Rate calibration - too high flow ratio

igiannakas, 2023-08-02:
### Bambu Studio Version

1.7.1

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Attempted flow rate calibration with:
1. Polyterra Red PLA
2. It is a matte filament
3. Bambu Lab Hot Plate with liquid glue
4. Select Calibration -> Flow Rate -> Auto Calibration

The resulting Flow Ratio was 1.02. The default for the filament is 0.98 and by doing a manual flow calibration the optimal value is at 0.965.

The below picture is with the calibrated value from the automatic flow rate calibration:
[image]

The below picture is with the manually calibrated value.
[image]

And a comparison - left automatic, right manual:

[image]

It is clear that the automatic process measures this as too high, resulting in significant over extrusion

### Actual results

Over extrusion on the automatically calibrated print

### Expected results

Right extrusion

### Project file & Debug log uploads

[ExtrusionMultiplier-Flow_Calibration.3mf.zip](https://github.com/bambulab/BambuStudio/files/12245103/ExtrusionMultiplier-Flow_Calibration.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

dragon-eyes-bbl, 2023-08-03:
Hi @igiannakas ,
Thanks for testing out this function.
It looks like the measurement is a bit off. I will need more info to investigate the problem on the machine. 
Could you upload the log from the machine to our cs and provide me with a ticket number?

Besides, did you have any chances to test another filament that has different color?

Thanks

igiannakas, 2023-08-03:
Yeap it happened with grey Polyterra too. I’ll test with some more today but it seems that it struggles to find the right value.

[image]

log uploaded on ticket US230803265001

the above is the last print in the logs

dragon-eyes-bbl, 2023-08-03:
@igiannakas 
Thank you very much.
I have received the logs, it is very helpful. I am investigating the problem.

dragon-eyes-bbl, 2023-08-03:
@igiannakas  Before I can locate the issue, you may try cleaning your microlidar lens with IPA. It may help.  Thanks.

igiannakas, 2023-08-03:
> @igiannakas Before I can locate the issue, you may try cleaning your microlidar lens with IPA. It may help. Thanks.

Doing so now. Will revert back in a few moments

igiannakas, 2023-08-03:
> @igiannakas Before I can locate the issue, you may try cleaning your microlidar lens with IPA. It may help. Thanks.

Cleaned the LiDAR with IPA. Tried calibrating twice with the Red and Grey polyterra PLA filaments

**Red filament:**
**First run:** 0.992 
**Second run:** 0.995

**Grey filament:** 
**First run:** 1.061
**Second run:** 1.059

Looks like it is still reading high. I have updated the ticket with the latest logs.

ThoughtfulDev, 2023-08-03:
Sounds a little related to #2144 
I too created a ticket. Im curious as to whats the issue here :) Thanks guys for the quick answers

igiannakas, 2023-08-03:
> Sounds a little related to #2144 I too created a ticket. Im curious as to whats the issue here :) Thanks guys for the quick answers

Yeap indeed. Also my k value is too high but don’t open another issue as I saw yours.

Alexx40in, 2023-09-30:
I have the same problem. The automatically calibrated flow is significantly greater than the manually calibrated flow. By about 15%

dragon-eyes-bbl, 2023-10-07:
> I have the same problem. The automatically calibrated flow is significantly greater than the manually calibrated flow. By about 15%

Hi @Alexx40in 
Automatic calibration requires an even surface finish for lidar to measure. Some filaments with reflective, glossy or sparkling finish are known that not compatible with this function. You could upgrade your printer firmware to the latest one which has been improved for capability.

closed 2024-01-23 by igiannakas