bambulab/BambuStudio #2153: Calibration result management for multiple printers

guitarshreddar, 2023-08-02:
The new auto-calibration addition and ability to save presets is extremely appreciated, to the extent that I have switched back to Bambu Studio (from OrcaSlicer) for the first time since 3 months ago. My problem is that, while the auto-calibration is great (and perfectly suited for individual users), I have 2 X1Cs. The current layout of the calibration makes it such that auto-calibrations are saved on a printer-specific basis. 

_This isn't a "make or break" issue, more of a nice to have. I believe that it is worth creating this post because I don't believe it will be extremely difficult to implement (based on existing data fields saved to printers within accounts and the ability to already store variables, it should just be a relatively easy add) and feel that it will be a major factor for multi-bambu printer users choosing a slicer software._

**Describe the solution you'd like**
What I'm looking for is for calibration variables (K value and flow rate) to be saved by nozzle size => by account vs the current approach of being saved by nozzle size => by printer => by account. Just take out the printer part in the "manage result" portion and aggregate by nozzle sizes by account.

zhimin-zeng-bambulab, 2023-08-04:
@guitarshreddar 
The k value is not only relative to the viscosity of the melted filament , but also relative to the melt zone diameter , nozzle diameter , nozzle length . As there is tolerance in the manufacturing , the melt zone diameter , nozzle diameter , nozzle length will not perfectly match the design , so the final compensation parameter for an specific filament will be different on different machines , and for now there is no a good solution to decouple these factors. 
So, currently k value is still related to the machine.

guitarshreddar, 2023-08-04:
I didn't even think about the manufacturing tolerance aspect. I realize now that it makes sense that even nozzles labeled the same diameter would have slight variances. Thank you for your clarification!

closed 2023-08-04 by guitarshreddar