bambulab/BambuStudio #2155: network plugin

maxalford, 2023-08-02:
### Bambu Studio Version

1.7.0.65

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11 pro

### Additional system information

 Operating System: Windows 11 Pro 64-bit (10.0, Build 22621) (22621.ni_release.220506-1250)
                 Language: English (Regional Setting: English)
      System Manufacturer: ASUS
             System Model: System Product Name
                     BIOS: 2004 (type: UEFI)
                Processor: 12th Gen Intel(R) Core(TM) i9-12900K (24 CPUs), ~3.2GHz
                   Memory: 131072MB RAM
      Available OS Memory: 130814MB RAM
                Page File: 21699MB used, 128569MB available
              Windows Dir: C:\WINDOWS
          DirectX Version: DirectX 12
 Card name: AMD Radeon RX 6900 XT
        Manufacturer: Advanced Micro Devices, Inc.
           Chip type: AMD Radeon Graphics Processor (0x73AF)
            DAC type: Internal DAC(400MHz)
         Device Type: Full Device

### Printer

bambu lab X1 carbon

### How to reproduce

setup bambu lab 1.7.1.65

### Actual results

failed install

### Expected results

when loading version 1.7.1.65 it does not load the network plug in

### Project file & Debug log uploads

none available

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-08-03:
@maxalford 

hello, could you upload the log file here for us to check?

please make sure not open any other bambustudio instance when install

jason-e-gross, 2023-08-03:
I am getting this same issue with the new version.    1.7.0.65 worked fine,  1.7.1.62 is now throwing errors saying it can't install the plugin.

bambu123, 2023-08-03:
@jason-e-gross @maxalford 
Please refer #1748

maxalford, 2023-08-03:
> 

#1748 nothing to do with this problem

jason-e-gross, 2023-08-03:
I have Teams installed, seems weird it'd have a hold of the plug in.  I've rebooted and will try exiting teams when it's back up, then retry.

jason-e-gross, 2023-08-03:
> @jason-e-gross @maxalford Please refer #1748

That worked for me (I'm assuming quitting Teams).   After my reboot, and making sure Teams wasn't installed - the plugin installed successfully just fine.   Thanks @bambu123

maxalford, 2023-08-03:
> > @jason-e-gross @maxalford Please refer #1748
> 
> That worked for me (I'm assuming quitting Teams). After my reboot, and making sure Teams wasn't installed - the plugin installed successfully just fine. Thanks @bambu123

i don't have teams installed

bambu123, 2023-08-03:
> > > @jason-e-gross @maxalford Please refer #1748
> > 
> > 
> > That worked for me (I'm assuming quitting Teams). After my reboot, and making sure Teams wasn't installed - the plugin installed successfully just fine. Thanks @bambu123
> 
> i don't have teams installed

Some other apps such as Skype, the Video Players may also load BambuSource.dll.

maxalford, 2023-08-03:
> Some other apps such as Skype, the Video Players may also load BambuSource.dll.

have uninstalled skype and the problem still exists

bambu123, 2023-08-03:
> have uninstalled skype and the problem still exists

Then you should delete the old plugins manually. 
You can locate the plugin folder for the menu "Help > Open Configuartion Folder".
Maybe some apps unknown load the BambuSource.dll.

maxalford, 2023-08-03:
> BambuSource.dll.

you want me to delete BambuSource.dll from the plugin directory

bambu123, 2023-08-03:
> > BambuSource.dll.
> 
> you want me to delete BambuSource.dll from the plugin directory

Yes.

jason-e-gross, 2023-08-03:
Yes, because something has a file lock on it, which is what's preventing the new plug in to install.  You could get a tool/app that lists who has file handles on it, or Brite force delete it so the studio can install the new version.

jason-e-gross, 2023-08-03:
Brute force, note Brite. Sorry.

maxalford, 2023-08-03:
it was being used by agent

maxalford, 2023-08-03:
stopped use by agent and program installed ok thanks for your help

bambu123, 2023-08-03:
> stopped use by agent and program installed ok thanks for your help

By the way, what is the agent?

maxalford, 2023-08-03:
https://en.wikipedia.org/wiki/Microsoft_Agent

bambu123, 2023-08-03:
> https://en.wikipedia.org/wiki/Microsoft_Agent

Got it. Thank you!

RyanEwen, 2023-08-04:
Just ran into this issue with today's hotfix release. Wasn't able to install b/c the file wasn't writable. Had tried rebooting, uninstalling, reinstalling, etc but nothing worked until closing Teams. 

If I told it to ignore the error and finishing installing, then Bambu Studio would launch, but the plugin would fail to install, likely for the very same reason. So there are 2 symptoms to the same issue I think.

lanewei120, 2023-08-05:
> Just ran into this issue with today's hotfix release. Wasn't able to install b/c the file wasn't writable. Had tried rebooting, uninstalling, reinstalling, etc but nothing worked until closing [...]

@RyanEwen 
thanks very much for doing this
"rebooting" means reboot the system, or just reboot bambu studio?

if install the new version on an old version
it may still fail due to the same reason
if install new version based on this new version, the situation should be much better

maxalford, 2023-08-05:
go to C:\Users\xxxxxxx\AppData\Roaming\BambuStudio\plugins
delete : BambuSource.dll

it won't delete but it will show you whatever program has it open just go to task manager processes and end the task holding it open and then it will be able to be deleted once you do that install the bambu networking app and all will be good there might be more than one program holding the BambuSource.dll open so do the same and end task

RyanEwen, 2023-08-05:
> @RyanEwen thanks very much for doing this "rebooting" means reboot the system, or just reboot bambu studio?

Reboot Windows.

bambu123, 2023-08-08:
Starting from version 1.7, there is a bug that registers redundant components when registering BambuSource, causing third-party software to load BambuSource. Starting from version 1.7.2, redundant registrations are deleted during installation, but installing a new version may still fail because BambuSource is loaded, so you still need to manually remove BambuSource and close the associated third-party app. Redundant registrations should be removed after successful installation of the new version and plugins.

closed 2023-08-08 by bambu123

Merlwynd, 2023-08-14:
> Starting from version 1.7, there is a bug that registers redundant components when registering BambuSource, causing third-party software to load BambuSource. Starting from version 1.7.2, redundant [...]

Note that during installation of BambuStudio 1.7.3, Skype still blocked overwriting BambuSource.dll. I had to close down Skype before the network plugin would install....