bambulab/BambuStudio #2156: Auto Tree Support Stopped Working

mcdavidr, 2023-08-02:
### Bambu Studio Version

1.7.1.62

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu X1C

### How to reproduce

import model, in prepare select auto tree support, in preview no support is generated. manual tree still works
I have sliced in 1.6 and last public beta sucessfully
[image]

 

### Actual results

no supports generated
[image]

### Expected results

tree supports generated

### Project file & Debug log uploads

[telaio_in_tubi-TH.zip](https://github.com/bambulab/BambuStudio/files/12245512/telaio_in_tubi-TH.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

mcdavidr, 2023-08-03:
I just tried on file that it worked on previously and it works? There must be an issue with the file I uploaded. I printed with manual tree supports, it printed but the support interface did not work correctly.

mcdavidr, 2023-08-04:
I think it’s a model specific issue

closed 2023-08-04 by mcdavidr

tangketan, 2023-08-07:
The supports are not generated because you enabled this option "Support critical regions only". Critical regions are sharp tails and long cantilevers for now, which your model doesn't have.
[image]

mcdavidr, 2023-08-07:
Thank you for clarifying this.Ron McDavid