bambulab/BambuStudio #2157: Flow Dynamics Auto Calibration - Invalid nozzle diameter

MwC-Trexx, 2023-08-03:
### Bambu Studio Version

1.7.1.62

### Where is the application from?

Bambu Lab github releases

### OS version

Mac OSX 13.5

### Additional system information

N/A

### Printer

Bambu X1 Carbon v1.06.00.00

### How to reproduce

Click on Calibration
Click on Flow Dynamics
Choose Auto-Calibration
Select 0.4 mm Nozzle & Cool Plate
Click Calibrate

### Actual results

https://github.com/bambulab/BambuStudio/assets/7128246/5ed34825-9a59-4920-a180-7aa1d89a9264

Error - Invalid Nozzle Diameter

### Expected results

Auto-Calibration begins

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12246354/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

drewzh, 2023-08-03:
I'm also experiencing this with the 0.6mm nozzle.

dmacpro91, 2023-08-03:
Also experiencing this regardless of what nozzle size I select.

MarcelLeon20, 2023-08-04:
I have the same issue.

paulmaksimovich, 2023-08-05:
Just ran into this too, 0.2mm nozzle.

MarcelLeon20, 2023-08-05:
Hey i think i found a solution. You have to set the right nozzle diameter on the printer display. In my case i have a X1C and there it is under maintenance --> accessoires --> nozzle. I hope i could help.

Edzi11, 2023-08-06:
Had Same issue when trying a .2 nozzle. had success with .4 nozzle prior to hotfix update

zhimin-zeng-bambulab, 2023-08-07:
We currently only support 0.4mm nozzle, and we will gradually improve the calibration of other nozzle diameters in the future.

mbukosky, 2023-08-08:
Thanks for the information, it would be a better user experience if the other nozzle sizes were greyed out and there was a tooltip that said this would be released in a later update.

kohones, 2023-08-19:
Version 1.7.4.52 - problem still exists.

klack, 2023-08-19:
The problem still exists

zhimin-zeng-bambulab, 2023-08-21:
At present, when we calibrate, the nozzle diameter selected by Studio needs to be the same as the nozzle diameter set by the printer.  Is your printer nozzle 0.6 mm?

counterdrone, 2023-08-21:
How do you mean "set by the printer"? I have a .4 nozzle installed, have selected such in the prep page, and yet when I go to calibrate it defaults to .6 and when I change to .4 get "invalid nozzle diameter." Where does this need to be changed at the printer level? Thanks.

MarcelLeon20, 2023-08-21:
Hi together. I think you have to set the diameter noozzle on the printer. I have a X1C and there it is under maintenance.

counterdrone, 2023-08-21:
Yup it's under "accessories" on the printer touchscreen thanks.

MwC-Trexx, 2023-08-22:
Appears to be fixed in 1.7.4.   You need to make sure the Nozzle size physically configured on the printer matches the nozzle size specified in the calibration screen in the slicer.

 I had forgotten that I had changed it on the printer a while ago for doing some TPU printing at 0.6mm.  After I switched the hotend back to a 0.4mm nozzle, I forgot to set the printer's back configuration accordingly.  

I am not sure what (if any) difference that made as the slicer was set to 0.4mm and seemed to print just fine.  It would be nice if there was an option in the slicer to update/configure the printer's nozzle configuration.

zhimin-zeng-bambulab, 2023-08-28:
Could you verify if this problem still exists in the latest version(V1.7.4).

jaxer, 2023-08-29:
I was not able to select nozzle 0.2 on printer as there is no such option there for me (only 0.4, 0.6, 0.8 but not 0.2). and i got this error when trying to calibrate :(

zhimin-zeng-bambulab, 2023-08-30:
@jaxer  What type of machine is yours and what is the firmware version?

jaxer, 2023-08-30:
> @jaxer  What type of machine is yours and what is the firmware version?

[image]

Fw 01.06.03.00

zhimin-zeng-bambulab, 2023-08-31:
[image]

@jaxer  There is no 0.2 for this type of nozzle, you can switch the nozzle type

jaxer, 2023-09-01:
Thanks @zhimin-zeng-bambulab that makes sense. I wish i understood that both of those are clickable earlier 🤦 
I was able to select correct nozzle there, have not tried calibrating again yet as long job is in progress.

dale-in-NM, 2023-09-05:
Ok, if Bambu only supports the 0.4 nozzle: does that mean I have to install my 0.4 nozzle, calibrate and then re-install my 0.8 nozzle.  Or, do I calibrate with my 0.8 nozzle as if it was 0.4mm?

zhimin-zeng-bambulab, 2023-09-06:
The latest firmware version can support nozzles other than 0.4 mm, just make sure that the nozzle installed on the printer is consistent with the diameter of the nozzle selected by Studio.

dale-in-NM, 2023-09-07:
Sorry, I have tried starting all over, rebooting the system, changing to 0.4 and then back to 0.8, changing filament, tried auto calibrate and manual calibrate but every time I get the “Invalid nozzle diameter” calibration error.

 

Please help.

 

Dale Shell

klack, 2023-09-14:
This feels like it isn't fixed.  I shouldn't have to set the nozzle in the slicer and on the printer.

dale-in-NM, 2023-09-15:
My problem ended up being that I selected 0.8 in Studio but didn’t change the selection on the printer. Sent from my iPhone

zhimin-zeng-bambulab, 2023-10-13:
Could you have a try in Studio V1.7.7 with latest firmware version

zhimin-zeng-bambulab, 2023-11-23:
This issue can now be verified in V1.8.0.57, just ensure that the nozzle diameters of Studio and printer are consistent.

closed 2023-11-23 by zhimin-zeng-bambulab

Modusmundi, 2024-01-04:
The error messages you get still aren't really helpful and don't really direct you to change nozzle at the physical printer; I guess I know now you need to change it in multiple spots, but the documentation here isn't great and I only found out my issue through Google landing me on this Github issue.  On latest firmware (1,07.01.00) and latest slicer.