bambulab/BambuStudio #2158: Slower purge and wipe after a pause/resume

thunderpipesusa, 2023-08-03:
1. Add a pause print in any projects layer 
2. Slice and send job to x1c
3.  When printer pauses hit resume.  
Alternatively just pause a print that has started using the pause button on the screen.

watch the extruder purge and wipe… 
This purges and wipes at a different speed than when the printer starts job sent to it.  The initial purge and wipe works perfectly.  Also during a color change when a purge and wipe happens Nothing stays on nozzle.  

The much faster purge and wipe after a pause can cause the overly fast and large diameter filament purge to not wipe off correctly and drag it to the print which can attach there and cause a print failure.  This has happened to me many times and I see many posts about this on the bambu owner Facebook group. I have cleaned the nozzle and even tried a new wiper even though the original wiper has only a weeks worth of prints on it.  Still happens only on a pause.  Please make the pause purge and wipe the same as the color change speed of purge and wipe and the initial before printing purge and wipe which are much slower and easier paced.

I expect the resume function to allow the purge and wipe be at the same speed as the color change purge and wipe.  We have never had any filament pulled to the plate or an object being printed during the slower purge and wipe speed at a color change or before actually starts printing the first layer.  I don’t have a file to upload as this happens with even the on board internal files that came with our printers.  Choose any file and print… then pause, then resume, watch the purge amount and speed, and watch the wipe speed… compare that to a color change purge and wipe and the initial after LiDAR reads purge and wipe before actual first job layer… they are different and the pause/resume purge and wipe is causing issues at times bringing the quick purged filament back to the model after the quick wipe because it has flung it around and caused it to stick somewhere else on the nozzle.

RyanEwen, 2023-08-04:
I, too, have noticed that the nozzle wipe after resuming a print is not always effective. Often times filament is dragged onto the part.

thunderpipesusa, 2023-08-04:
Glad it’s not just me, I pause on every print for prep (magnets etc) and it’s hit or miss whether something is dragged to the part

skl111, 2023-11-05:
This happens with me as well

chaley01, 2024-04-20:
Has any work been done on this? Hoping to bump this back up as a priority if not.

Englund81, 2024-07-04:
This is an issue for me as well. Added a 👍 to the initial post in hope of getting it prioritised.

ArshansGithub, 2024-12-26:
Any fix? Still experiencing this

brahmus66, 2025-02-09:
Every time I pause a print, a line of purged filament is dragged onto the bed after resuming and creates a spaghetti mess. EVERY TIME! It doesn't do it after a color change. This it totally unacceptable and should be fixed immediately. It is clearly caused by different purge/wipe speeds and parameters than a color change. Please fix it soon.