bambulab/BambuStudio #2160: Allow filament and process profiles to be applied to multiple printer models

CoryCharlton, 2023-08-03:
**Is your feature request related to a problem? Please describe.**
I often share the same filament and process settings across both my P1P and X1C. In the current implementation I have to manually update the settings for both machines if I decide to change a setting.

**Describe the solution you'd like**
An option to instruct the profile to be applicable/available to multiple printer models.

**Describe alternatives you've considered**
Manually copying the settings. 
Writing my own application to "sync" the settings from one machines profile to the other model(s).

**Additional context**
None.

KD7LRJ, 2023-08-03:
I have 2 X1C and 4 P1P and I would like to have this ability also. I usually just send the X1C files directly to the P1P machines and click past the printer mismatch pop-up message and they work fine.

coreyward, 2023-12-26:
+1 — I have an X1C, an A1, and an A1 Mini, and I want to use the same type of filament (custom settings) across all three devices. For some reason Bambu Studio makes the preset in a way that is machine specific, meaning I cannot possibly operate a print farm with a custom filament without a ton of extra work.