bambulab/BambuStudio #2161: Toggle to inform Bambu Studio of a P1P that was upgraded to a P1S via the official upgrade kit

adam2104, 2023-08-03:
I recently upgraded my P1P to a P1S using the official upgrade kit sold in the Bambu Store. However, Bambu Studio still sees the printer as a P1P (as expected). This prevents me from using the P1S filament/printing presets which have different temperature and cooling properties than the P1P's presets.

**Describe the solution you'd like**
It would be great if there was some way to inform Bambu Studio that my P1P has been upgraded to P1S capabilities so as to unlock the P1S printing and filament presets / profiles. Perhaps this could be done with a setting in the machine profile?

**Describe alternatives you've considered**
I tried just using the P1S profile in Bambu Studio but it displays a warning about incorrect g-code when attempting to send a print to the printer. I also tried creating user-made presets with the parameters matching the system presets but this is tedious, prone to error, and won't receive automatic updates like the stock presets.

RyanEwen, 2023-08-04:
Would be even better if there was the option to flash the P1S firmware. Alternatively a toggle in the P1P firmware settings menu to mark it as upgraded to P1S. 

This way not every app and third-party tool (things that use MQTT like Home Assistant P1Touch, etc) can automatically determine P1P vs P1S rather than each having some kind of workaround of their own

naldoxable, 2023-08-11:
up

d-komarov, 2023-08-15:
As a temporary workaround to prevent warning about incorrect g-code I just change the `model_id` from **C12** to **C11** in resources\profiles\BBL\machine\Bambu Lab P1S.json.

lilyharuna, 2023-09-09:
+1. I really want this feature.

IMiteBeWrong, 2023-09-11:
Bump. 

Would really appreciate this as well; I was a little surprised that it wasn't already implemented by the time the first public shipments of the upgrade kit went out.  

The kit includes front plate with the P1S logo on it. I may be silly to assume, but this somewhat suggests to me that once the kit is installed, Bambu Labs considers it a P1S (And obviously the rest of the parts included appears to make it identical).

Still waiting on mine to ship, should go out by the end of this month... Hopefully this will be added by then.

Thank you!

skuridin, 2023-09-23:
Front panel of the printer literally says P1S, but Bambu Studio thinks otherwise. This creates unnecessary confusion. 

Would be great to hear **some** feedback from the maintainers.

tsmith35, 2023-09-25:
The biggest issue that the misidentification causes has to do with process profiles. The process profiles have two basic identifiers: "X1C" or "P1P".

This means that any profiles for the X1, X1C or P1S will happily work with all of them. But if a P1P is selected as the device, none of the user process profiles will appear. They're just not available.

I've found two ways to deal with the situation: one is to duplicate all of the user process profiles and meticulously change them to have "P1P" in the appropriate places. The other is to manually choose "P1S" as the printer type. And, of course, acknowledge the incorrect machine type for... Every. Single. Job.

Ideally, with the creation of the P1P-to-P1S kit, users should be given a means of selecting which printer identity is desired. It can be on the printer, it can be in the account or it can be in the software. Resorting to text-editing json files to work around a programming error is not a good solution. It's almost as bad as spending hundreds to update an MK3S+ to an MK3.9 -- if I am upgrading, I don't want to end up with a 99% upgrade.

RyanEwen, 2023-09-27:
I received 2 new P1S's today to go along side 2 P1P's which I will be upgrading to P1S's, and I can already see how annoying it is going to be if I can't have the P1P's identified as P1S's. 

None of my custom P1P filament profiles or quality profiles can be used with my new P1S's. It's going to be frustrating to duplicate all of my custom profiles, and the frustration will continue forever if I have to maintain both sets of profiles whenever I make changes to one in the future. 

**Please allow the P1P to be flashed with P1S firmware** (and vice-versa in case of downgrade/mistake). Or at least create a printer setting to mark upgraded P1P's as P1S's from the LCD menu.

IAmTheFuzz1, 2023-10-18:
+1
Official upgrade parts should mean official firmware compatibility.

benr916, 2023-10-23:
Waiting on this too!

gmonteir, 2023-10-24:
Three months and nothing... this should be an easy fix, what gives?

marczai, 2023-10-24:
+1
Want it too. 
Waiting atm on the website to buy the kit.

rickmills, 2023-10-30:
+1 Very disapointing that such a trivial change hasn't been added to either handy or studio, or heck even just a toggle on the printer UI itself.

SuperRonanCraft, 2023-11-02:
definite deal breaker, waiting on a firmware update before buying the upgrade kit, not wanting to click confirm each time I want to print the same model on multiple printers

Zakary64, 2023-11-10:
+1 
I created an account just to bump this issue because I recently upgraded my P1P to P1S and it's annoying to press the confirmation that I am using the "wrong" gcode for my actually upgraded printer. Please make my devices see it as it is

Jacc79, 2023-11-11:
I was thinking of buying the part that contains the firmware. I don't know which part has the firmware. I can only guess, but I think it's on the AP card. I was thinking of buying a P1S AP katy for P1P and swapping it. However, I did not find an AP card among the accessories at Bambu.

tsmith35, 2023-11-11:
Just contact support and ask then to send you an invoice for the board. Pay and they'll send it. I think it's about $50 for the P1x AP board.

rickmills, 2023-11-11:
It'll be the mainboard, but that's not a realistic solution as it'll be one of the most expensive parts of the whole printer, not to mention this really is a simple case of a firmware fix, and an update to bambu's cloud api to support pushing out a notice that the printer has been upgraded so that its serial can be marked as a P1S in their DB instead of a P1P, or at very least marked as an upgraded P1P so it fixes all the weird issues you get when trying to use P1S presets.

QingZhangBambu, 2023-11-13:
I will talk to the concerned staff about that.

tsmith35, 2023-11-13:
> I will talk to the concerned staff about that.

Thank you, thank you, thank you!

Honestly, a simple toggle in the UI would do it. There's really no harm in describing a P1P as a P1S, or vice-versa, since they functionally operate the same.

I could add fans and an enclosure to my P1P and have a P1S clone, or I could take a P1S and open it up (remove glass and possibly replace back cover) for ease of maintenance if using it for PLA and PETG on a print farm. They're still operationally equivalent.

QingZhangBambu, 2023-11-13:
Here is some information I've learned, you may want to know now:
1. Studio does not prohibit you from printing P1S profile, P1S's profile is compatible. this warning is aimed to tell you that the printer type is different from the type set on the firmware to avoid someone having the wrong type selection. Click "confirm", and keep print.
2. This warning depends on the machine type set on the firmware. For now, Changing machine type in firmware may cause other issues on firmware.

QingZhangBambu, 2023-11-13:
@sodachen22 please track this issue, and consider a fix in the feature.

tsmith35, 2023-11-14:
> 1. Studio does not prohibit you from printing P1S profile, P1S's profile is compatible. this warning is aimed to tell you that the printer type is different from the type set on the firmware to avoid someone having the wrong type selection. Click "confirm", and keep print.

Thank you for the response. While the requirement for clicking "confirm" is not a major issue, it is a major annoyance to anyone that has purchased the P1S upgrade kit. It is a constant reminder that they didn't purchase a "real" P1S (even if the upgrade covers everything but the firmware), thereby devaluing what would otherwise be a wonderful upgrade and making purchasers of the upgrade regret said purchase.

"Factory" P1S = $699
P1P ($599) + P1S "upgrade kit" ($150) = $749

It's insulting to pay $50 beyond the price of a "factory" P1S, then be constantly reminded that one really only owns a P1P. An upgrade to a different model designation should be a "true" upgrade, i.e., the P1P should become a true P1S. The confirmation dialog serves no purpose, as the user is certainly quite aware of what they own and doesn't need to be unintentionally shamed for having upgraded their Bambu Lab product every single time they submit a print job.

If I accidentally sent a P1P print job to a P1S, what bad things might happen? Absolutely nothing. What about sending a P1S print job to a P1P? Will that cause major problems? No. It will work exactly the same as expected. If I'm not smart enough to tell that my printer has an enclosure (or not), I should probably find another hobby.

> 2. This warning depends on the machine type set on the firmware. For now, Changing machine type in firmware may cause other issues on firmware.

I have serious doubts about two identical printers, differentiated only by a handful of non-critical components, suffering any issues from sharing the same firmware. The difference between a P1P and a P1S is:
1. Body panels and taller back panel
2. A door and top glass
3. Three (3) additional fans
4. A cable chain with reinforced cable
5. A front bezel for appearance only
6. A carbon air filter
7. Three (3) pieces of PTFE tubing
8. A few bits of data in firmware

There is nothing there would be adversely affected by replacing the letter "P" with the letter "S" in the firmware. If the firmware works on a P1P, it will work 100% on a P1S. If it works on a P1S, it will work 100% on a P1P. It shouldn't take an act of congress to implement a minor feature ("Do you have a P1P or a P1S?") in firmware. It's just not that difficult. It's not.

Jacc79, 2023-11-14:
It is simply incomprehensible why they don't let you upgrade P1P to P1S once they sell the upgrade package.

RyanEwen, 2023-11-14:
> Thank you for the response. While the requirement for clicking "confirm" is not a major issue, it is a major annoyance to anyone that has purchased the P1S upgrade kit. It is a constant reminder [...]

This is well put.

It was painful to migrate my print quality profiles and filament profiles from P1P to P1S, but I eventually did it. Now every time I send a print and am reminded that the machine type doesn't match, I feel like my upgraded machines are lower class citizens, or fake. And yet they cost more than my new P1Ss. 

I was tempted to request purchase links for P1S AP boards but that would be a waste of money and effort for hardware I already have on hand and installed in the machine. 

If modifying the firmware to allow switching between P1P and P1S is not ideal or Bambu, then I think they should setup a system similar to how beta firmware is obtained. For beta firmware, we can send our serials and then beta firmware will be pushed to our machines. Let us send in our P1P serial numbers, and push out P1S firmware to our machines.

As it is now, I feel bitterness and resent, since the upgrade packages are not true upgrades. Not to mention they don't include the chamber fan duct, and the door gasket is the wrong color.

danicolome, 2023-11-19:
I completely agree, if you update the p1p to p1s it is in all areas, including the firmware. Please Bambu, we want a solution

czsquizer, 2023-11-23:
Hello, is there any progress on it, please?

djgringoboy2003, 2023-12-02:
This needs bumped, simply annoying...

JakubComi, 2023-12-04:
+1 I really want to buy the kit but this is really dumb to have it like that. And also, the maker world printing will force the P1P instead of P1S… Please, please 🙏  make it that we can upgrade via OTA like beta FW or make some toggle, to make it work flawlessly as an ACTUAL P1S. 
Thanks!

capsel22, 2023-12-05:
> > Studio does not prohibit you from printing P1S profile, P1S's profile is compatible. this warning is aimed to tell you that the printer type is different from the type set on the firmware to avoid someone having the wrong type selection. Click "confirm", and keep print.
> 
> It's not just that. If you want to use manual calibration in Bambu Studio, your created profile will be created for P1P, not P1S and you need to manually migrate it.

I didn't realise this either. How do you migrate profile from P1P to P1S?

Phil-Barker, 2023-12-05:
It's honestly very annoying on a lot of levels

I had upgraded my P1P pretty much from day 1 to have a motherboard fan and a printed enclosure. You have to add GCode to start and stop the fan as it's not included in the P1P profile. Every time there's an update you don't get updated GCode because you're using a custom one, so I had to do a file compare and manually migrate the changes over.

As the P1S upgrade kit physically changes the hardware to match the P1S, it's enclosed, the fans need to run, you absolutely SHOULD use the P1S settings, but then get a stupid warning every time you print that you're printing to the wrong machine.

This seems like such a simple fix but 5 months later I doubt we're ever going to see support for it

tsmith35, 2023-12-05:
> > It's not just that. If you want to use manual calibration in Bambu Studio, your created profile will be created for P1P, not P1S and you need to manually migrate it.
> 
> I didn't realise this either. How do you migrate profile from P1P to P1S?

Basically, you have to save the values created for your P1P (even if it's been upgraded, it's still a P1P to Bambu Lab) and manually enter them into your so-called "P1S" profile.

Trust me, it's just easier to pretend you still have a P1P, since Bambu Studio and OrcaSlicer both see your upgrade-P1S as just a P1P. Enjoy the enclosure and treat your upgrade-P1S like it's still a P1P, and you won't have to acknowledge the popup (which appears every single time you try to print) that you're trying to print to a P1S and your printer reports that it's a P1P.

I'll bet it would take exactly two bytes in the firmware to eliminate that popup, but the only way around it is to sell your so-called P1S and buy a real P1S.

JakubComi, 2023-12-06:
And now they rised the price… Now it isnt worth it: https://eu.store.bambulab.com/en-cz/products/p1p-to-p1s-upgrade-kit

kyzxyz, 2023-12-06:
https://forum.bambulab.com/t/firmware-upgrade-path-for-p1p-to-p1s/29413

Everyone, please post in the forum

itsonlymedan, 2023-12-06:
An application for users to fill out to upgrade to P1S firmware with prerequisites and then an OTA firmware update. This is annoying

JakubComi, 2023-12-06:
> > And now they have rised the price… Now it isnt worth it: https://eu.store.bambulab.com/en-cz/products/p1p-to-p1s-upgrade-kit
> 
> Yeah I checked the product few minutes ago out of curiosity and was amazed by the price increase. I think it's about 60-70% more expensive now. Even with the current discount (-25%) that's running, the price is higher than it used to be. Hopefully nobody is actually buying it for that price. I'm glad I got mine while the price was still somewhat reasonable.

You are lucky one 😁 I didn’t and probably won’t get it for that price… But I am kinda sad, this is hillarious…

naldoxable, 2023-12-06:
True, in Italy I bought it at 164€ and now its about 215€ with 25% discount.

JakubComi, 2023-12-06:
Wow, I want to do so, but I didn’t because it was sold out :( Do you know some resellers that have it for lower price that bambu now? In EU

naldoxable, 2023-12-06:
> Wow, I want to do so, but I didn’t because it was sold out :( Do you know some resellers that have it for lower price that bambu now? In EU

Try here: [](https://bambulab.com/en-eu/dealer)

tokamac, 2023-12-06:
> Wow, I want to do so, but I didn’t because it was sold out :( Do you know some resellers that have it for lower price that bambu now? In EU

Same, I've been trying to buy this kit for weeks to no avail when it was at 160 euros, even if I thought it was expensive enough already. I even subscribed to the totally useless "NOTIFY ME WHEN AVAILABLE" notification. Totally useless as they lie with that button, they NEVER send the notification email when the item is back in stock, making you miss the product over and over again (I was not aware of that despicable practice at first, someone opened my eyes me a few days ago, and alas it's true). 

Now, with that insane price increase I give up. I cannot even resell my P1P to buy a P1S instead without losing a huge amount of money as I was among the first customers to buy the P1P at over 800 € back then. Yes, that is to say more expensive that the P1S when it appeared afterward! Today the P1P + P1S kit would cost me over 1,000 euros! That's how Bambu rewards its most loyal customers. 
#PissedOff

QingZhangBambu, 2023-12-07:
it's on our plan list. We are investigating how to implement it. Please be patient and feel free to tell us what you want.

tsmith35, 2023-12-07:
> it's on our plan list. We are investigating how to implement it. Please be patient and feel free to tell us what you want.

From my earlier post on your forum:

The need for a way to select printer identity (P1P or P1S) should be provided to all P1x owners for the sake of accommodating both those who have upgraded their P1Ps piecemeal and those who chose to get the kit.

Going further, it would be prudent to eliminate the existing parallel profile structures that exist, one for the P1P only, and the other for the X1C / P1S only. While having separate profile sets made sense when only the P1P and X1C existed, the P1S blurred the distinction between the two, with the P1S adopting the X1C profile set. Since the P1S is just a P1P with an enclosure and a handful of non-critical components, and since the P1S works just fine using the X1C profile set, I question the need to maintain a separate profile set for the P1P only. Consolidating the two profile sets into one single set would simplify both software development and customer profile maintenance. A true win-win for everyone involved.

Let’s face it: maintaining two sets of profiles is a pain in the rear. Not only do I have to maintain process and filament profiles for numerous configurations, but I also have to maintain an identical set of those profiles for the P1P, since the P1P can’t see X1C profiles, and the X1C/P1S can’t see the P1P profiles. Every modification must be made twice. It’s a pain. There’s no point anymore.

Jruiz10, 2023-12-12:
This is an easy fix that should not be ignored especially since they offered the upgrade. It feels half done and needs to be fixed.

threadstone83, 2023-12-14:
Have we been heard? Here is an excerpt from the new Bambu Studio 1.8.2:

**No popup while sending to print if the P1P installed P1P to P1S upgrade kit**
In this version, if the P1P to P1S upgrade kit is installed, when sending for printing, if the machine preset is P1P, a compatibility reminder will pop up. Otherwise, if it is P1S, the compatibility reminder will no longer pop up. The functionality will be available after the firmware update to version 01.05.00.00 (the public beta version, will come out soon)

capsel22, 2023-12-14:
rejoice fellow upgraders

tsmith35, 2023-12-14:
> Have we been heard? Here is an excerpt from the new Bambu Studio 1.8.2:
>
> **No popup while sending to print if the P1P installed P1P to P1S upgrade kit** In this version, if the P1P to P1S upgrade [...]

So now I wonder how this will be accomplished. A cloud-based settings toggle? Selector on the printer display? Slicer-based toggle? Hmm. Guess I'll have to be patient.

marko-hologram, 2023-12-28:
Writing "+1" comments doesn't help anyone and is just spam. Please people, upvote the original comment to let the Bambu team know this is something that you want and support as an idea. Same applies to other ideas/requests as well.

Btw, there is probably no need to upvote this more because judging by the Bambu Studio release notes, they are planning to address this with Bambu Studio + firmware update.

wwwdata, 2023-12-28:
Guys it is fixed with the latest beta firmware. You can then just toggle under Accessories settings that you upgraded the printer and also that you installed the hardened steel nozzle then it will not warn you anymore when you select the P1S profile or harder materials :)

tsmith35, 2023-12-28:
The toggle for P1P/P1S is in the Accessories screen, as @wwwdata said.

[image]

[image]

tsmith35, 2023-12-28:
> Btw, there is probably no need to upvote this more because judging by the Bambu Studio release notes, they are planning to address this with Bambu Studio + firmware update.

It's actually already available. From within Bambu Handy, select your printer, then tap the "3-bar" at the top right corner. In the page that opens, select "Firmware Version". You'll see a prompt at the top about joining the open beta. Do it, then you'll have access to the latest firmware. You may need to power-cycle your printer to allow it to see the firmware update.

When upgrading firmware, I follow this sequence:
1. Install firmware
2. Power-cycle printer
3. Run full calibration
4. Power-cycle printer
5. Go have fun

You may not need to do all this, but I've flashed firmware on literally thousands of devices, so I've learned that following a stable procedure can save a lot of time and grief.

kyzxyz, 2023-12-29:
It's good to have some closure on this. Thanks to those who made some noise about this issue, and thanks to the Bambu team for making this happen.

Phil-Barker, 2024-01-02:
I enabled the beta firmware on the 28th, 5 days ago. Power cycled the printer multiple times. Logged out and in. But no firmware update is appearing on either the printer or bambu handy 😞 

Really want to try this out, don't know what I'm doing wrong.

kyzxyz, 2024-01-02:
Looks like official update (non beta) is being rolled out, be patient!

Phil-Barker, 2024-01-02:
Got it now. Don't want to spam this thread now it's resolved but credit where credit is due. 
This firmware update is a big QOL improvement!

The P1P upgraded enclosure toggle is fantastic, the motor noise cancellation is awesome but being able to view the SD card remotely and re-print directly from the desktop app is a game changer for me.

Thank you so much bambu lab - smashed it out the park with this one!

QingZhangBambu, 2024-01-10:
Attention, please: for the latest version the flow ratio result would be logged to P1S
[image]

capsel22, 2024-01-10:
> Attention, please: for the latest version the flow ratio result would be logged to P1S [...]

doesnt work as per https://github.com/bambulab/BambuStudio/issues/3344

Oniandr, 2024-01-16:
> Attention, please: for the latest version the flow ratio result would be logged to P1S [...]

Hi, bambulab p1s have a regional lock, or only p1p have regional lock?

voshond, 2024-01-16:
This software toggle etc seems like an unnecessarily complex solution. Instead of this approach, can't we simply flash P1S firmware to out P1Ps via Bambu Studio? This way all existing logic could be reused when it comes to device & related management.

Maybe you can add some attribute to the firmware, that this was a P1P once, that the flasher knows how to set, so we can also avoid having to have P1S custom firmware for upgraded users. 

Unless this is not possible due to some hardware differences, but I thought there are none or electronically none.

tsmith35, 2024-02-15:
@adam2104 I think this issue has been corrected, so you may be safe to mark it completed.

closed 2024-02-16 by adam2104