bambulab/BambuStudio #2162: Since update to V1.7.1.62 (macOS) the videostream very often hung up after few seconds/minutes

sladexos, 2023-08-04:
### Bambu Studio Version

1.7.1.62

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Ventura 13.5

### Additional system information

CPU: 2,6 GHz 6-Core Intel Core i7
RAM: 16 GB 2400 MHz DDR4
Graphic: Radeon Pro 555X 4 GB / Intel UHD Graphics 630 1536 MB

### Printer

Bambu Lab P1P

### How to reproduce

1. Start Videostream

### Actual results

Videostream hung up

### Expected results

Videostream don't hung up

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zISKo666, 2023-08-05:
Hey,

I have the same issue on Windows 11 and on my Android phone with the Bambu app.
Maybe some sort of network issue!?

Kind regards
zISKo

bambu123, 2023-08-07:
Please wait for P1P firmware update.

bambu123, 2023-08-07:
From BambuStudio 1.7, we support remote live view connection to P1P. But the P1P 1.3 version has problems with remote connect. The P1P 1.4 version is coming.

zISKo666, 2023-08-07:
No change with P1P @ 1.4
Ever so often the stream to Bambu Studio and the Bambu app freezes.

bambu123, 2023-08-09:
> No change with P1P @ 1.4 Ever so often the stream to Bambu Studio and the Bambu app freezes.

Sorry to hear this, please send me the new logs.

lanewei120, 2023-08-23:
@zISKo666 
do you still have this issue with the newest firmware and studio?

if yes
could you help tp upload the log files to us?
you can follow the guide https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

bambu123, 2024-02-06:
It is a memory leak problem in P1P. The new firmware 1.5 fixed it.