bambulab/BambuStudio #2163: New release does not remember pre-print options.

werewolfenn, 2023-08-04:
### Bambu Studio Version

1.7.1.62

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro- 64 bit

### Additional system information

enough

### Printer

X1CC, fully updated firmware

### How to reproduce

Get ready to send job to print.
Box for print options opens, check boxes show up, choosing Bed Level, AMS, etc.

Flow Dynamics will not stay checked. Must be rechecked each print.  

### Actual results

[](url)
[image]

### Expected results

[image]

### Project file & Debug log uploads

[sample for error.zip](https://github.com/bambulab/BambuStudio/files/12256429/sample.for.error.zip)
[debug_network_Thu_Aug_03_22_14_17.zip](https://github.com/bambulab/BambuStudio/files/12256444/debug_network_Thu_Aug_03_22_14_17.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2023-08-04:
@werewolfenn 
Hi.
what's type of your printer bed

werewolfenn, 2023-08-04:
It's the textured PEI, by Bambu.  I see now that setting I was looking to leave checked is a new option / feature.  I see it works differently, so I was mistaken.

walterwongbbl, 2023-08-07:
@werewolfenn 
Flow Dynamics Calibration is not selected by default when you are using textured PEI.

closed 2023-08-23 by walterwongbbl