bambulab/BambuStudio #2164: No resume button after layer or spaghetti warning

werewolfenn, 2023-08-04:
### Bambu Studio Version

17162

### Where is the application from?

Bambu Lab Official website

### OS version

Win 10 pro - 64 bit

### Additional system information

plenty

### Printer

X1C

### How to reproduce

[image]

### Actual results

Cannot resume after a stoppage due to layer or spaghetti alert.   There is no button to resume like there had been.    Printers also running latest firmware.

### Expected results

Used to see PLAY button to resume after a pause or error.   Not now.

### Project file & Debug log uploads

[sample for error.zip](https://github.com/bambulab/BambuStudio/files/12256466/sample.for.error.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

werewolfenn, 2023-08-04:
No RESUME option at machine screen either or in Handy app.

KageKeeper, 2023-08-15:
I am experiencing this exact same issue.

walterwongbbl, 2023-08-16:
Hi. @werewolfenn 
We need to confirm the message here. 
There is indeed no resume button here

werewolfenn, 2023-08-16:
Correct.  After I had done the firmware update that came out around 2 weeks ago, RESUME was no longer an option after a spaghetti fail.   I rolled the firmware back, and I see RESUME once again.   I have not  checked if newer firmware is available or not since then.

walterwongbbl, 2023-08-23:
@werewolfenn 
Hi.
Are you saying that if you roll back the firmware version, the printer screen will display the "RESUME" button?

werewolfenn, 2023-08-23:
Yes Walter,  when I rolled back, RESUME returned.    I'm now running the most recent FW in one of mine, and the second most recent in my others.  I haven't had this issue come up yet, to see.

SaltWei, 2023-08-24:
https://github.com/bambulab/BambuStudio/issues/2351#issuecomment-1687926185

closed 2023-08-24 by SaltWei