bambulab/BambuStudio #2165: After autom. Flow Dynamics Calibration - Values are not reflected in Bambu studio

Panda631, 2023-08-04:
### Bambu Studio Version

1.7.1.62

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Additional system information

AMD Ryzen 7 3800X 8-Core Processor                3.89 GHz
16,0 GB

### Printer

X1C & 2xAMS

### How to reproduce

1. Start Flow Dynamics: auto calibration and Flow Rate: auto-calibration
2. Restart Printer and Bambu Studio (power off - power on)
3. Goto Device Tab. AMS settings show K 0.000 

### Actual results

With this settings the print goes totaly wrong

### Expected results

The calculated K value should be read from the printer and shown in the AMS screen.

### Project file & Debug log uploads

No files can be uploaded.

Message appears: 
"Soemthing went wrong, and we can't process that file. Try again"

### Checklist of files to include

- [X] Log file
- [X] Project file

MagicMilamber, 2023-08-07:
same Setup (ryzen7, x1c 2ams, Win11, BS 1.7.2 ), same Problem for me

MagicMilamber, 2023-08-10:
There is no change to do. Calibaration result is an Error and it tell us nothing is saves.

Panda631, 2023-08-10:
> @Panda631 @MagicMilamber After you start the automatic calibration, Did you save calibration results after the calibration is completed?

Sure, the results hast been stored with the News name

zhimin-zeng-bambulab, 2023-08-17:
could you have a try for  v1.7.4

Panda631, 2023-08-17:
Checked it with the new version 1.7.4.

Unfortunately no improvement for this topic.
Each time the printer and bambustudio is restarted, you must select the PA profile for each filement manually in the device settings.

[image]

zhimin-zeng-bambulab, 2023-09-01:
@Panda631  I am sorry that we do not support recording the calibration results corresponding to each ams slot after printer is restarted.

Panda631, 2023-09-01:
You don't must apologize for that. You have in your hands to change it. 😉

Panda631, 2024-08-06:
Problem is still present and still anoying!
It can't be rocket scince to remember the last settings.