bambulab/BambuStudio #2166: No Thumbnail on Klipper/Marlin Printers

Robnex, 2023-08-04:
### Bambu Studio Version

1.7.1.62

### Where is the application from?

Bambu Lab Official website

### OS version

Windows10

### Printer

Anycubic Vyper / Kingroon KP3S 

### How to reproduce

**Option 1:**
1. Slice a box
2. Upload to the printer
3. No gcode

**Option 2:** (see Image)
1.  Slice a box
2. Export as Gcode
3. Open Gcode with editor
4. See no thumbnail section in gcode File

### Actual results

I cant see any Thumbnail on Mainsail, Fluid or KlipperScreen.

### Expected results

I Should see a thumbnail on Mainsail, Fluid or KlipperScreen.
Or it should be a thumbnail section in gcode file.

### Project file & Debug log uploads

File: 
Left = Bambu Studio / Right = Orca Slicer

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Robnex, 2023-08-04:
Also There is no Thumbnail Option on Printer Settings like in Orca Slicer.
[image]

chuckkay, 2023-08-08:
also no before layer change and after layer change in gcode (just layer change gcode)

SaltWei, 2023-08-16:
@XunZhangBambu Please add the thumbnail settings and picture back when printer is 3-rd party. Thanks

slynn1324, 2023-08-17:
It looks like the part that actually injects the thumbnail into the gcode was just commented out at some point...  I was able to add it back with the below patch.  Probably should be expanded to actually make it configurable like the screenshots above... 

```
diff --git a/src/libslic3r/GCode.cpp b/src/libslic3r/GCode.cpp
index 22377156..e93b7992 100644
--- a/src/libslic3r/GCode.cpp
+++ b/src/libslic3r/GCode.cpp
@@ -1563,9 +1563,11 @@ void GCode::_do_export(Print& print, GCodeOutputStream &file, ThumbnailsGenerato
     }
 
     //BBS: add plate id into thumbnail render logic
-    //DoExport::export_thumbnails_to_file(thumbnail_cb, print.get_plate_index(), THUMBNAIL_SIZE,
-    //    [&file](const char* sz) { file.write(sz); },
-    //    [&print]() { print.throw_if_canceled(); });
+    if ( ! print.is_BBL_Printer() ){
+        DoExport::export_thumbnails_to_file(thumbnail_cb, print.get_plate_index(), THUMBNAIL_SIZE,
+            [&file](const char* sz) { file.write(sz); },
+            [&print]() { print.throw_if_canceled(); });
+    }
 
 
     // Write some terse information on the slicing parameters.
diff --git a/src/libslic3r/GCode/ThumbnailData.hpp b/src/libslic3r/GCode/ThumbnailData.hpp
index fef48070..71a68187 100644
--- a/src/libslic3r/GCode/ThumbnailData.hpp
+++ b/src/libslic3r/GCode/ThumbnailData.hpp
@@ -8,7 +8,7 @@
 namespace Slic3r {
 
 //BBS: thumbnail_size in gcode file
-static std::vector<Vec2d> THUMBNAIL_SIZE = { Vec2d(50, 50) };
+static std::vector<Vec2d> THUMBNAIL_SIZE = { Vec2d(300, 300) };
 
 struct ThumbnailData
 {
```

slynn1324, 2023-08-18:
see PR  #2333

SaltWei, 2023-08-19:
This is original function of prusa and useless for banbuprinter，so we removed it. We will add it back for 3rd party printer

SaltWei, 2023-08-19:
Thanks for the PR. We will have a review

closed 2023-08-29 by XunZhangBambu

freedomanonyme, 2024-01-05:
Just waiting to have the function back. Thanks!

alpo011, 2024-01-22:
Function still isn't available in 1.8.4. Please add back.

AvengingFrog, 2024-01-25:
> It looks like the part that actually injects the thumbnail into the gcode was just commented out at some point... I was able to add it back with the below patch. Probably should be expanded to [...]

So, it looks like they are not going to add the function back anytime soon. Can anyone tell me what config file this is contained in so that I can re-enable this myself? TIA!

alpo011, 2024-01-25:
Doesn't seem like it. I have an open request too.

XunZhangBambu, 2024-01-26:
> So, it looks like they are not going to add the function back anytime soon. Can anyone tell me what config file this is contained in so that I can re-enable this myself? TIA!

@AvengingFrog Hi,thumbnail information has been encoded in base64 and written into the GCode file, and it can be correctly viewed after decoding. However, different printers have different parsing method, leading to a failure to display.