bambulab/BambuStudio #2168: [Issue] Exclude object doesn't update time to finish

BambuLabDiscord, 2023-08-04:
When i exclude an object from a 7 hour print. it still says it will complete in 7 hours, even though i just remove 50% of the print.

SaltWei, 2023-08-05:
The accurate estimated time and printing progress is written into gcode when slicing by M73 command. Impossible to be changed. Studio can do nothing.
When some models printing fails, there is nothing more important than continuing to finish the remaining part. Time is no longer important. It can only be less than full printing.

SaltWei, 2023-10-07:
This is not important and also impossible to be enhanced by current time estimation method in studio.
Won't do.

Unless the time estimation is done by printer itself. And every time user remove objects, the printer check all remaining gcode to estimate a new accurate time. VERY big feature and impossible to be finished in this generation printer as well. :(